google/xls #1222: build should not require external library depedencies

proppy, 2023-12-11:
As originally pointed in #1, it looks like the libtinfo5 implicit dependency is still present:
```
external/llvm_toolchain_llvm/bin/clang: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
```

We should find a way to make this self contained to the bazel workspace.

proppy, 2023-12-11:
This requirement can apparently be turned out with `LLVM_ENABLE_TERMINFO=OFF`.

cdleary, 2023-12-15:
@proppy This commit looks like it removed the need for terminfo? https://github.com/llvm/llvm-project/commit/7b5d6cd7fcac2b9e06338f2497fed9039360924a

cdleary, 2023-12-15:
Oh, but it looks like it comes from the binary toolchain we download for hermetic build -- not the bazel llvm rules -- in that snippet you posted.

proppy, 2023-12-15:
oh you're right, so maybe we simply need to use a more recent version:
- bump https://github.com/google/xls/blob/main/WORKSPACE#L24-L31 to  https://github.com/grailbio/bazel-toolchain/releases/tag/0.10.3
- bump https://github.com/google/xls/blob/main/WORKSPACE#L39-L42 to https://github.com/llvm/llvm-project/releases/tag/llvmorg-17.0.6

proppy, 2026-10-28:
I think we should be able to clean those up from the README/Dockerfile now we switch to managing llvm toolchain deps w/ bazel mod.