google/xls #1224: Optimization opportunity: pull out branches from constant sub-expressions to constant-evaluate each branch

hzeller, 2023-12-11:
Consider the following `foo.x`

```rust
fn top(b: bool) -> u32 {
  u32:3 + (if b {u32:1} else {u32:2})
}
```

Processing
```python
IR_CONVERTER=bazel-bin/xls/dslx/ir_convert/ir_converter_main
IR_OPT=bazel-bin/xls/tools/opt_main
CODEGEN=bazel-bin/xls/tools/codegen_main
${IR_CONVERTER} foo.x | ${IR_OPT} --top=__foo__top - | ${CODEGEN} --module_name=top --generator=combinational -
```

### Result

The result leaves the expression as-is so leaves the detection of the constant addition to the synthesis-tool:

```verilog
module top(
  input wire b,
  output wire [31:0] out
);
  wire [2:0] add_48;
  assign add_48 = {1'h0, b ? 2'h1 : 2'h2} + 3'h3;
  assign out = {29'h0000_0000, add_48};
endmodule
```

### Expected result:

It would be good to recognize the opportunity that all the leaves of the expressions are constant expressions. That would allow to pull out the `if`-condition and then pre-process the resulting two constant expression additions `(3 + 1)` and `(3 + 2)` and choose these with the branches, so something like:

```verilog
module top(
  input wire b,
  output wire [31:0] out
);
  wire [2:0] add_48;
  assign add_48 = b ? 3'h4 : 3'h5;
  assign out = {29'h0000_0000, add_48};
endmodule
```

This would then set the stage for potentially better follow-up optimizations. Like in this case, the existing xls optimizer would then see the following simplification:

```verilog
module top(
  input wire b,
  output wire [31:0] out
);
  assign out = {31'h0000_0002, ~b};
endmodule
```

hzeller, 2024-01-23:
(edit: changed name of function from `top` to `foo` as the former is a keyword now)

allight, 2024-01-23:
While waiting for some ACLs to propogate I got most of this done. I'll upload soonish.

Some improvements.

closed 2024-01-24 by copybara-service[bot]