google/xls #1232: INTERNAL: BytecodeEmitter when calling `uadd_with_overflow`

proppy, 2023-12-18:
The following snippet:
```
%%dslx --top=uadd8 --generator=combinational
import std

fn uadd8(a: u8, b: u8) -> u8 {
  let (sum, overflow) = std::uadd_with_overflow(a, b);
  sum
}

#[test]
fn adders_test() {
  assert_eq(uadd8(u8:1, u8:2), u8:3);
}
```
product this error:
```
E1218 08:06:26.424927    4108 command_line_utils.cc:45] Could not extract a textual position from error message: INTERNAL: BytecodeEmitter could not find slot or binding for name: V @ /content/conda-env/share/xls/xls/dslx/interpreter_main.runfiles/com_google_xls/xls/dslx/stdlib/std.x:654:6-654:7 stack: 0x59e6a13218c9: xls::dslx::BytecodeEmitter::HandleNameRefInternal()
0x59e6a1320ab1: xls::dslx::BytecodeEmitter::HandleNameRef()
0x59e6a139e5db: xls::dslx::NameRef::AcceptExpr()
0x59e6a131f0b0: xls::dslx::BytecodeEmitter::HandleInvocation()
0x59e6a13a04b8: xls::dslx::Invocation::AcceptExpr()
0x59e6a130b0b2: xls::dslx::BytecodeEmitter::EmitExpression()
0x59e6a1232f5f: xls::dslx::internal::(anonymous namespace)::EagerlyPopulateParametricEnvMap()
0x59e6a12320a2: xls::dslx::internal::FunctionInstantiator::Instantiate()
0x59e6a122f3c7: xls::dslx::InstantiateFunction()
0x59e6a11d90bf: xls::dslx::InstantiateParametricFunction()
0x59e6a11ca01f: xls::dslx::CheckInvocation()
0x59e6a11d80b8: std::__1::__function::__func<>::operator()()
0x59e6a1225201: xls::dslx::(anonymous namespace)::DeduceInstantiation()
0x59e6a1201749: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x59e6a13a048b: xls::dslx::Invocation::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a1202c45: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x59e6a13a172b: xls::dslx::Let::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f6bb4: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x59e6a13a152b: xls::dslx::Statement::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11fb9b8: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x59e6a13a0cbb: xls::dslx::Block::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a11c5728: xls::dslx::CheckFunction()
0x59e6a11d0db7: xls::dslx::CheckModule()
0x59e6a115df6e: xls::dslx::TypecheckModule()
0x59e6a115d5b3: xls::dslx::ParseAndTypecheck()
0x59e6a115472d: xls::dslx::ParseAndTest()
0x59e69f87966a: main
0x79d41f74fd90: [unknown]
: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: BytecodeEmitter could not find slot or binding for name: V @ /content/conda-env/share/xls/xls/dslx/interpreter_main.runfiles/com_google_xls/xls/dslx/stdlib/std.x:654:6-654:7 stack: 0x59e6a13218c9: xls::dslx::BytecodeEmitter::HandleNameRefInternal()
0x59e6a1320ab1: xls::dslx::BytecodeEmitter::HandleNameRef()
0x59e6a139e5db: xls::dslx::NameRef::AcceptExpr()
0x59e6a131f0b0: xls::dslx::BytecodeEmitter::HandleInvocation()
0x59e6a13a04b8: xls::dslx::Invocation::AcceptExpr()
0x59e6a130b0b2: xls::dslx::BytecodeEmitter::EmitExpression()
0x59e6a1232f5f: xls::dslx::internal::(anonymous namespace)::EagerlyPopulateParametricEnvMap()
0x59e6a12320a2: xls::dslx::internal::FunctionInstantiator::Instantiate()
0x59e6a122f3c7: xls::dslx::InstantiateFunction()
0x59e6a11d90bf: xls::dslx::InstantiateParametricFunction()
0x59e6a11ca01f: xls::dslx::CheckInvocation()
0x59e6a11d80b8: std::__1::__function::__func<>::operator()()
0x59e6a1225201: xls::dslx::(anonymous namespace)::DeduceInstantiation()
0x59e6a1201749: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x59e6a13a048b: xls::dslx::Invocation::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a1202c45: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x59e6a13a172b: xls::dslx::Let::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f6bb4: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x59e6a13a152b: xls::dslx::Statement::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11fb9b8: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x59e6a13a0cbb: xls::dslx::Block::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a11c5728: xls::dslx::CheckFunction()
0x59e6a11d0db7: xls::dslx::CheckModule()
0x59e6a115df6e: xls::dslx::TypecheckModule()
0x59e6a115d5b3: xls::dslx::ParseAndTypecheck()
0x59e6a115472d: xls::dslx::ParseAndTest()
0x59e69f87966a: main
0x79d41f74fd90: [unknown]

Error: INTERNAL: BytecodeEmitter could not find slot or binding for name: V @ /content/conda-env/share/xls/xls/dslx/interpreter_main.runfiles/com_google_xls/xls/dslx/stdlib/std.x:654:6-654:7 stack: 0x59e6a13218c9: xls::dslx::BytecodeEmitter::HandleNameRefInternal()
0x59e6a1320ab1: xls::dslx::BytecodeEmitter::HandleNameRef()
0x59e6a139e5db: xls::dslx::NameRef::AcceptExpr()
0x59e6a131f0b0: xls::dslx::BytecodeEmitter::HandleInvocation()
0x59e6a13a04b8: xls::dslx::Invocation::AcceptExpr()
0x59e6a130b0b2: xls::dslx::BytecodeEmitter::EmitExpression()
0x59e6a1232f5f: xls::dslx::internal::(anonymous namespace)::EagerlyPopulateParametricEnvMap()
0x59e6a12320a2: xls::dslx::internal::FunctionInstantiator::Instantiate()
0x59e6a122f3c7: xls::dslx::InstantiateFunction()
0x59e6a11d90bf: xls::dslx::InstantiateParametricFunction()
0x59e6a11ca01f: xls::dslx::CheckInvocation()
0x59e6a11d80b8: std::__1::__function::__func<>::operator()()
0x59e6a1225201: xls::dslx::(anonymous namespace)::DeduceInstantiation()
0x59e6a1201749: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x59e6a13a048b: xls::dslx::Invocation::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a1202c45: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x59e6a13a172b: xls::dslx::Let::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f6bb4: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x59e6a13a152b: xls::dslx::Statement::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11fb9b8: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x59e6a13a0cbb: xls::dslx::Block::Accept()
0x59e6a11f37eb: xls::dslx::Deduce()
0x59e6a11d7e15: std::__1::__function::__func<>::operator()()
0x59e6a1239d98: xls::dslx::DeduceCtx::Deduce()
0x59e6a11f3cc2: xls::dslx::DeduceAndResolve()
0x59e6a11c5728: xls::dslx::CheckFunction()
0x59e6a11d0db7: xls::dslx::CheckModule()
0x59e6a115df6e: xls::dslx::TypecheckModule()
0x59e6a115d5b3: xls::dslx::ParseAndTypecheck()
0x59e6a115472d: xls::dslx::ParseAndTest()
0x59e69f87966a: main
0x79d41f74fd90: [unknown]
```

See the following notebook for an easy repro:
https://colab.research.google.com/gist/proppy/0a353c1cff2cca397ddabc08c9fcc15d/xls-playground-sandbox.ipynb

proppy, 2023-12-18:
as @hongted pointed setting the type parameter explicitly:
```
import std

fn uadd8(a: u8, b: u8) -> u8 {
  let (_, sum) = std::uadd_with_overflow<u32:8>(a, b);
  sum
}

#[test]
fn adders_test() {
  assert_eq(uadd8(u8:1, u8:2), u8:3);
}
```
make the internal error go away, but the tooling should probably return a friendlier error message.

proppy, 2024-01-19:
I get a similar error when using derived parametrics:
```
fn unpack3<A_SZ: u32, B_SZ: u32, C_SZ: u32, R_SZ: u32 = {A_SZ+B_SZ+C_SZ}>(n: uN[R_SZ]) -> (uN[A_SZ], uN[B_SZ], uN[C_SZ]) {
  (n[(B_SZ+C_SZ)+:uN[A_SZ]], n[C_SZ+:uN[B_SZ]], n[0+:uN[C_SZ]])
}

#[test]
fn unpack3_test() {
 let (a, b, c): (u3, u3, u2) = unpack3(u8:0b101_010_11);
 assert_eq(a, u3:0b101);
 assert_eq(b, u3:0b010);
 assert_eq(c, u2:0b11);
}
```
which is related to some alternative slicing syntax we were discussing with @hzeller today.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/dslx/type_system_v2/inference_table_converter_impl.cc:2696-2705` catches uninferred parametric bindings (`Could not infer parametric(s): V`) during typechecking before `BytecodeEmitter` runs, returning a span-annotated `TypeInferenceErrorStatus` instead of an internal error. (Closed by Gemini)

closed 2026-11-14 by allight