google/xls #1255: xls::dslx::ConvertOneFunction() fails to convert parametric functions

rw1nkler, 2024-01-09:
The `xls::dslx::ConvertOneFunction` [docstring](https://github.com/google/xls/blob/1da08a2e3146a0708719a273119f24d56a8e2015/xls/dslx/ir_convert/ir_converter.h#L67) states that the `parametric_env` parameter can be used to provide bindings to use during conversion if the function is parametric. However, the function fails to convert parametric functions even when the correct `parametric_env` is passed to the conversion function.

Here is the [branch](https://github.com/antmicro/xls/tree/53590-convert-one-function) showing this problem.
One can run the example using:
```
$ bazel run  //xls/examples:convert_one_function
INFO: Analyzed target //xls/examples:convert_one_function (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //xls/examples:convert_one_function up-to-date:
  bazel-bin/xls/examples/convert_one_function
INFO: Elapsed time: 0.124s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Running command line: bazel-bin/xls/examples/convert_one_function
Unable to convert one function: INTERNAL: Not enough symbolic bindings to convert function: foo; need {WIDTH} got {}
```

cdleary, 2024-01-20:
This was more involved than I thought it was going to be so I stalled on the first attempt, LMK how important it is for making forward progress or if you have a workaround, thanks!

rw1nkler, 2024-02-14:
Yes, we have a workaround - we create a function that wraps the parametic function with concrete parameters.