google/xls #1261: python3: cannot execute binary file on ARM Mac

spurserh, 2024-01-13:
A new user trying to set up XLS on an ARM Mac got an error "python3: cannot execute binary file" from the bazel build. The path shown is to the x64 python binary. 

The user says "I moved the python39_x86-64 folder out of the way,  and made that folder name symlink to the python_aarch64 folder -> works".

proppy, 2024-01-13:
since we switched to hermetic toolchain in https://github.com/google/xls/commit/eb2fb6d81523b09fbd77b63b88381b1ad6c74850 `rules_python` should be able to select the right python variant to download https://github.com/bazelbuild/rules_python/blob/a79b66a76eeee4a8c7b967cde80e9ded82411631/python/versions.bzl#L191-L201 according to the target platform.

@spurserh do have the full log, I suspect it could be a rogue target that doesn't use the hermetic toolchain interpreter with
```
load("@python39//:defs.bzl", python_interpreter_target = "interpreter")
```

cdleary, 2024-01-23:
@spurserh Also note to help with these we'd probably need more step by step instructions on how the environment got set up, because the nightly OS X build is looking ok: https://github.com/google/xls/actions/runs/7622196412/job/20759833772 Notably that's not running any test targets though.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): XLS migrated to Bzlmod in `MODULE.bazel` with multi-platform `@rules_python` (`version = "1.9.0"`) and `@llvm` (`targets = ["AArch64", "X86"]`), resolving the legacy x86_64 Python binary execution failure on ARM64 macOS. (Closed by Gemini)

closed 2026-11-14 by allight