google/xls #1262: XLS on Mac: org_theopenroadproject_asap7sc7p5t_28/license (Not a directory)

spurserh, 2024-01-13:
A new user is running into this error. He hypothesizes that it is due to Mac OS being case insensitive in file names, as there is a LICENSE file in that directory.

QuantamHD, 2024-01-16:
Which version of bazel are you using?

spurserh, 2024-01-16:
@QuantamHD  6.4.0

cdleary, 2024-01-23:
@spurserh Are the synthesis targets required for what the user is trying to do? Could try commenting them out for now if you just want XLS targets to generate RTL to start off.

We've been talking about whether we tag all the backend targets in the tree and have a config like no_backend that would prevent users trying to build backend targets.

spurserh, 2024-01-24:
He just wants to use xlscc, opt, codegen. Any possibility you could point me to where he should try commenting things out?

allight, 2026-11-14:
Closing this issue as obsolete (accepted by @allight): XLS migrated its OpenROAD ASAP7 dependency from `org_theopenroadproject_asap7sc7p5t_28` to `org_theopenroadproject_asap7_pdk_r1p7` / `asap7sc7p5t_27` (`xls/estimators/delay_model/BUILD:19`, `xls/build_rules/xls_oss_config_rules.bzl:66`), removing the `asap7sc7p5t_28/license` target. (Closed by Gemini)

closed 2026-11-14 by allight