google/xls #1299: Enhance tracing/logging

mczyz-antmicro, 2024-02-13:
1. In Verilog there are severity tasks: `$fatal`, `$error`, `$warning`, `$info` (c.f. IEEE Standard for SystemVerilog 1800-2017 Chapter 20.10 Severity tasks) and I feel that creating corresponding functions, e.g. `$fatal` -> `fatal!` would be intuitive and improve output readability. Here is an example, where code is significantly reduced:

* Current DSLX implementation:

```
if value_ok != value_err {
    trace_fmt!("Error occurred, expected {}, got {}", value_ok, value_err);
    assert_eq(value_ok, value_err);
} else {}
```

* I propose that the following syntax is enabled:

```
error!(value_ok == value_err,"This error occurred because of reasons");
```

* Alternatively, a more general assert statement:

```
assert(value_ok == value_err, "This error occurred because of reasons");
```
I believe that this functionality is already implemented in the [IR](https://google.github.io/xls/ir_semantics/#assert).

* This type of feature needs to be expanded in the Verilog form to:

```
assert (value_ok == value_err) $error("This error occurred because of reasons");
```

2. I also think that DSLX needs verbosity levels for the `trace` and/or `trace_fmt!` function. The implementation could define functions, which derive from existing `trace_fmt!()` function: `trace_fmt_log!()`, `trace_fmt_warn!()`, `trace_fmt_err!()`. This feature is useful for DSLX module development - if I run multiple tests, the output quickly becomes cluttered. The function prototypes could also use a global enum for the verbosity level, i.e. `trace_fmt!(DSLX_WARN, "Warning message")`.

3. I wish I could set a hierarchy level of traces, e.g. if I am debugging state of the top-level proc, then I would like to suppress output from 2nd level and following children of the top-level, but use traces from all 1st level children. This feature could work like a filter, e.g. `--set_trace_enable_level = 1` or by selecting proc names: `--set_trace_enable_name proc_name_*` 

#### Related issues

Issues related to this one:
* https://github.com/google/xls/issues/481
* https://github.com/google/xls/issues/651
* https://github.com/google/xls/issues/1071
* https://github.com/google/xls/issues/1082
* https://github.com/google/xls/issues/1236

meheff, 2024-05-01:
I presume fatal is expected to terminate the simulation/evaluation?

The fatal/error/info levels also matches the Google C++ logging library levels (and others I presume). Numeric verbosity levels in trace statements could correspond to VLOGging.

@grebe added verbosity levels to trace statements in the IR.  These should be surfaced in the DSLX. That could be the initial step.

A follow on could add fatal/error/info functionality.

mczyz-antmicro, 2024-05-06:
Yes, fatal should terminate evaluation

allight, 2024-10-02:
NB That the IR already supports verbosity levels in trace_fmt! but this isn't exposed to DSLX. Having support for this would be nice.

proppy, 2025-01-09:
@allight did we ever add the doc for https://github.com/google/xls/commit/0af76338b7e5ed64ed718c56af46f68d23b134cd ? (is it ready to be documented, i.e: plugged in the interpreter flags?)

proppy, 2025-01-09:
related: it would be nice to be able to dump leveled log into a structured format (json, proto) that could be more easily queried offline.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Verbosity-controlled tracing has been implemented via the `vtrace_fmt!(verbosity, fmt, ...)` builtin macro (`xls/dslx/frontend/parser.cc:3331` and `xls/dslx/frontend/builtins_metadata.cc:77`) and `--max_trace_verbosity` (`xls/codegen/trace_verbosity_pass.cc`). (Closed by Gemini)

closed 2026-11-14 by allight