google/xls #1303: IR Function converter for built in expressions assumes that immediate expression is an Array

hongted, 2024-02-14:
For FunctionConverter::HandleBuiltinOneHotSel (and similarly other selects) https://github.com/google/xls/blob/a2908f23437e63297a70422175e90373e6d08df3/xls/dslx/ir_convert/function_converter.cc#L2920-L2927  

The 1'th argument is assumged to be an Array Expression.  This may not be the case as the 1st argument could be another expression with an Array Type.  For example, in the below an Attr expression rather than an Array Expression is given.

```
import std;

pub struct Cases {
  data: bits[16][4],
}

pub fn SelectFromCases(sel: u4, cases: Cases) -> bits[16] {
  one_hot_sel(sel, cases.data)
}
```

hongted, 2024-02-15:
Fixed by https://github.com/google/xls/commit/84fa96f80084e9580af131007433ebb07d0c311e

cdleary, 2025-03-15:
@hongted This change doesn't make sense to me, why would we need to index an array bvalue just to put it back into an array? Any expression with a DSLX array type should have an associated array-typed BValue in IR conversion. cc @richmckeever

cdleary, 2025-03-15:
Oh I see -- at some point in the past it assumed it was always an ArrayExpr -- we should have Visit'd it and then Use'd it -- there should be no need to subsequently scalarize it and rebuild an array though. I'll make a change for that.

cdleary, 2025-03-15:
Nevermind -- even though that may have been true in the past, the fundamental issue (at least now) is more about the IR construction.

PrioritySel and OneHotSel take a `cases=[a, b, c]` style specification as a sequence of IR ops (that are used as operands). In the IR, `a, b, c` are given as variadic (i.e. 3 BValues) instead of as aggregates (i.e. one BValue that is array typed). So if we get something like an array parameter we want to do `one_hot_sel` or `priority_sel` on, we have to extract the elements to pass in this variadic fashion. We could make the IR ops take an array typed value, but usually we like to desugar aggregate values because it makes analysis more precise / makes it more direct to refer to ops in the op graph, so probably not much benefit in that.

Closing again now that I understand why the scalarization makes sense for all array-typed expressions. :-)

closed 2025-03-15 by cdleary