google/xls #1308: Use of a for in a parametric function in proc next function results in duplicate ConversionRecords

hongted, 2024-02-15:
IR conversion of the following dslx
```
fn count<IMPL: bool>(x:u32) -> u32 {
  for(i, v): (u32, u32) in u32:0..u32:10 {
   v 
  } (x)
}

fn main() -> u32 {
  count<false>(u32:0)
}

proc Counter {
 in_ch: chan<u32> in;
  out_ch: chan<u32> out;
 
  init {
    u32:0 
  } 

  config(in_ch: chan<u32> in, out_ch: chan<u32> out) {
    (in_ch, out_ch)
  }

  next(tok: token, state: u32) {
    let (tok, in_data) = recv(tok, in_ch);
    let x = count<false>(in_data);
    let next_state = state + x;
    let tok = send(tok, out_ch, next_state);

     next_state
  }
}
```

Results in an error message
```
INTERNAL: XLS_RET_CHECK failure (third_party/xls/ir/verifier.cc:1862) !name_set->contains(function_base->name()) Function/proc/block with name ____test__count__0_counted_for_0_body is not unique within package test
```

Suspect that https://github.com/google/xls/blob/e2302fd06afd603ebf8dca86cb38437887bddbb4/xls/dslx/ir_convert/extract_conversion_order.cc#L656 is not removing these duplicates due to them being partly parametric.

cdleary, 2024-02-21:
@hongted I think maybe you forgot to put "fixes" in that change description which would have automatically closed this -- it's fixed now, right?

hongted, 2024-02-22:
Yes, should be fixed.

closed 2024-02-22 by hongted