google/xls #1311: can't use std::upow in parametric expr

proppy, 2024-02-16:
trying to use `std::upow` in a parametric expr, like in the following example:
```
fn pow_param<N: u32, P: u32 = {std::upow(u32:2, N) - u32:1}>() -> uN[N] {
    P as uN[N]
}
```
result in the following error:
```
xls/dslx/stdlib/std.x:556:20-556:21
0554:         let result = if (n & uN[N]:1) == uN[N]:1 { result * p } else { result };
0555: 
0556:         (n >> 1, p * p, result)
~~~~~~~~~~~~~~~~~~~~~~~~~^ constexpr evaluation detected rollover in operation
0557:     }((n, p, result));
0558:     work.2
```
is that expected?

cdleary, 2024-02-17:
What was the value used for `N` in the instantiation?

We maybe can/should make the error message better but it's trying to tell you that `p * p` rolled over in the `u32` space which you may not have been expecting; i.e. `p * p` at infinite precision `bits[∞]` was not equal to the value when clamped to a `bits[32]`.

cdleary, 2024-04-03:
Closing as I don't think we have a reproducer.

closed 2024-04-03 by cdleary