google/xls #1312: ModuleNotFoundError: No module named 'google.api' in some tests

hzeller, 2024-02-18:
Testing on my  personal workstation [^1] there are a few tests that fail because they  all couldn't find some `google.api` Module.

I suspect some missing dependency somewhere which happens to work accidentally in other contexts ? Someone with some Python-fu might see immediately how to fix.

Stacktraces look like this:

```python
Executing tests from //xls/tools:codegen_main_test
-----------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/hzeller/.cache/bazel/_bazel_hzeller/27c61cc5bc4809ebb94526a6de7fd324/sandbox/linux-sandbox/39225/execroot/com_google_xls/bazel-out/k8-opt/bin/xls/tools/codegen_main_test.runfiles/com_google_xls/xls/tools/codegen_main_test.py", line 26, in <module>
    from xls.codegen import module_signature_pb2
  File "/home/hzeller/.cache/bazel/_bazel_hzeller/27c61cc5bc4809ebb94526a6de7fd324/sandbox/linux-sandbox/39225/execroot/com_google_xls/bazel-out/k8-opt/bin/xls/tools/codegen_main_test.runfiles/com_google_xls/xls/codegen/module_signature_pb2.py", line 15, in <module>
    from google.api import field_behavior_pb2 as google_dot_api_dot_field__behavior__pb2
ModuleNotFoundError: No module named 'google.api'
```

This affects these tests
  * //xls/synthesis:synthesis_utils_test
  * //xls/delay_model:op_module_generator_test
  * //xls/synthesis:timing_characterization_client_test
  * //xls/synthesis/yosys:yosys_server_test
  * //xls/synthesis:synthesis_server_test
  * //xls/synthesis/openroad:json_metrics_server_test
  * //xls/tools:codegen_main_test

[^1]: NixOS; had to slightly patch as the external llvm toolchain had to be commented out to use the local one and the Python 3.9 binary that was downloaded by bazel was not executable so was symlinking it to a local Python

hzeller, 2024-02-28:
After further investigation it indeed turns out that the root-cause is the `rules_python()` which downloads some non-hermetic Python binary using some shared libraries.
Closing, as for now, we only support a particular Ubuntu environment.

closed 2024-02-28 by hzeller

beekarthik, 2024-12-30:
Hey think I am running into a similar issue with pip_parse not working as intended for `googleapis-common-protos`. Any pointers on what may be the issue?

AFAIK rules_python should be providing a hermetic toolchain so system dependencies should not exist?