google/xls #1313: Change default `*` and `+` behavior to upcasting without loss of precision

iv2nl0b9v, 2024-02-20:
Current behavior:
```
fn multiply(a: s4, b: s4) -> s8 {
    (a * b) as s8
}
```
Gets translated to:
```
function automatic [3:0] smul4b_4b_x_4b (input reg [3:0] lhs, input reg [3:0] rhs);
  reg signed [3:0] signed_lhs;
  reg signed [3:0] signed_rhs;
  reg signed [3:0] signed_result;
  begin
    signed_lhs = $signed(lhs);
    signed_rhs = $signed(rhs);
    signed_result = signed_lhs * signed_rhs;
    smul4b_4b_x_4b = $unsigned(signed_result);
  end
endfunction
```

It is a potentially unexpected behavior: it is not consisten with Verilog as we can do `c = a * b` as long as `c` has the right width and it doesn't warn or fail.

allight, 2024-02-20:
Also note that we mostly try to copy rust which is pretty firmly in 'choice 1'. No implicit type-casts at all, operators return the same type of the inputs. Though rust also makes it a (runtime) error to overflow on some types so I guess an argument could be made for either being more 'rust-like'.

I wonder if maybe it would be useful to follow rust and have types which encode overflow behavior more explicitly. On the other hand that sort of operator-overloading can be kind of a mess.

cdleary, 2024-02-20:
As mentioned, the policy is to use rust style typing for the operator syntax.

If you want full multiplier width you can `import std` and do `std::umul` or `std::smul` which takes an X bit type and a Y bit type and returns a X+Y bit result.

https://google.github.io/xls/dslx_std/#stdmul

Going to close as the original post is asking for the default policy to change, and here we've addressed why it won't, but that there are other options including in the stdlib. Appreciate the filing!

closed 2024-02-20 by cdleary

hzeller, 2024-02-20:
FYI, there are languages that have annotations for things where the user explicitly allows an overflow.
https://docs.swift.org/swift-book/documentation/the-swift-programming-language/advancedoperators/#Overflow-Operators

It would make it somewhat explicit to the reader and would allow both choices, depending on the operator used.

But as said before above, in the software world (where rust is coming from), we lean towards choice 1, but we also only have a very limited set of integer types available.
In the hardware world we really count the bits. If we want an unsigned 9 bit value, we can, and choose it for a reason. If we add something to it or multiply we always get the wider result and explicitly cast it down if we want it truncated.

So to be less surprising to hardware developers it would be safer if operations would typically behave like choice 2 (automatically widen; `*` would result in 2x input size, `+` would result in +1 input size) so that then the user either needs to be explicit and cast down again. Or they use a truncating operation. Breaking the assumptions is thus explicit not silent. And it behaves like a hardware engineer would think about the problem.

This of course would not be backward compatible with existing DSLX and thus cause a lot of churn in existing code. Needed changes would be fairly mechanically and always safe though.