google/xls #1335: Support "overhanging combinational delay", e.g. on input parameters to a computation

cdleary, 2024-03-06:
I imagine this can be wired up to e.g. param nodes, reflecting them having delay in their arrival, in the SDC scheduling problem.

This is a thing people do try to avoid for their blocks (vs a cleanly timed flopped path), but it happens, and I think it's not too burdensome to support.

My napkin sketch is along the lines of (for simple pipelines):

```
fn f(x: MyInputStruct) -> MyOutputStruct {
  ...
}
```

`codegen_main --param_delay=x:10ps`

(And of course `sched_main` would really be the ideal binary in the command line but #724 is not done)

cdleary, 2024-03-06:
Note that right now you could work around the lack of this with margining, see `clock_margin_percent` on https://google.github.io/xls/codegen_options/#pipelining-and-scheduling-options

hongted, 2024-03-06:
Currently there is  --additional_input_delay_ps which adds a fixed delay to every receive op.  This suffers from xls/issues/575.

hongted, 2024-03-06:
Duplicate of https://github.com/google/xls/issues/575

closed 2024-03-06 by hongted