google/xls #1359: proc-scoped type aliases cannot be used with `zero!`

rw1nkler, 2024-03-27:
**Describe the bug**
https://github.com/google/xls/commit/91176922420dd0e79dfcd474171000a74d845cac introduced a syntax for defining convenient proc scoped aliases. 
However, when proc scoped aliases are used with `zero!` macro, the type cannot be resolved correctly.

**To Reproduce**

One can spot a problem by modifying the `init` function of the test proc from [proc_with_alias.x](https://github.com/google/xls/blob/main/xls/dslx/tests/proc_with_alias.x) in the following way:
```rust
proc P {
    type MyU32 = u32;
    s: chan<MyU32> out;

    config(s: chan<MyU32> out) { (s,) }

    init {
        type AnotherU32 = u32;

        //zero!<AnotherU32>() // this works correctly
        zero!<MyU32>() // this causes problems
    }

    next(tok: token, state: MyU32) {
        send(tok, s, state);
        let new_state = state + MyU32:1;
        new_state
    }
}
```

To run the test after modifications use:
```
bazel test //xls/dslx/tests:proc_with_alias_dslx_test
```
Here is the error message, that shows a problem with parsing the proc:
```
ERROR: /home/rwinkler/projects/xls/xls/dslx/tests/BUILD:1145:15: Parsing and type checking DSLX source files of target proc_with_alias_dslx failed: (Exit 255): bash failed: error executing command (from target //xls/dslx/tests:proc_with_alias_dslx) /bin/bash -c ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
xls/dslx/tests/proc_with_alias.x:25:15-25:20
0023:
0024:         //zero!<AnotherU32>() // this works correctly
0025:         zero!<MyU32>() // this causes problems
~~~~~~~~~~~~~~~~~~~~^---^ TypeMissingError: node: 0x5571295ea090 user: 0x5571295f7d10 internal error: AST node is missing a corresponding type: MyU32 0x5571295ea090 (kind: NameDef) defined @ xls/dslx/tests/proc_with_alias.x:16:10-16:15. This may be due to recursion, which is not supported.
0026:     }
0027:
Error parsing and type checking DSLX source file: xls/dslx/tests/proc_with_alias.x
Target //xls/dslx/tests:proc_with_alias_dslx_test failed to build
```

**Expected behavior**

`zero!` should be able to use proc scoped type aliases, and the snipped from the previous section should be parsed correctly.

**Additional context**

Related issues:
* https://github.com/google/xls/issues/1050
* https://github.com/google/xls/issues/1296

cdleary, 2024-03-27:
I forgot to update type inference to traverse to the new aliases, thanks for filing, should be a quick fix, and will try to think a bit about how to (pseudo-) fuzz new features like this.

closed 2024-03-27 by copybara-service[bot]

rw1nkler, 2024-03-28:
Thank you @cdleary for this fix! `zero!()` works correctly!

However, I have found one more problem with proc-scoped type aliases. It seems that, in some cases, there is a problem with the aliases that refer to parameters of the proc. The problem happened when I used proc parameters with a RAM message structures. 

Here is the test file showing the problem:
[top_level_alias_error.x.txt](https://github.com/google/xls/files/14790256/top_level_alias_error.x.txt)

And the error message displayed when I try to execute the test from the attached file:
```
ERROR: /home/rwinkler/projects/xls/xls/modules/zstd/BUILD:697:17: Parsing and type checking DSLX source files of target example_dslx failed: (Exit 255): bash failed: error executing command (from target //xls/modules/zstd:example_dslx) /bin/bash -c ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
E0328 14:20:16.266390       3 parametric_bind.cc:126] INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/parametric_bind.cc:126) ctx.parametric_binding_types.contains(pdim_name) Cannot bind RAM_ADDR_WIDTH : it has no associated type.
0x560aa5ccf5fa: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x560aa3ef8179: xabsl::StatusBuilder::operator absl::lts_20240116::Status()
0x560aa3ff0018: xls::dslx::ParametricBindTypeDim()
0x560aa3ff073b: xls::dslx::ParametricBind()
0x560aa3fe95e0: xls::dslx::internal::ParametricInstantiator::InstantiateOneArg()
0x560aa3fed4e1: xls::dslx::internal::StructInstantiator::Instantiate()
0x560aa3fe7ff8: xls::dslx::InstantiateStruct()
0x560aa3fb9cc0: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStructInstance()
0x560aa435641b: xls::dslx::StructInstance::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3fa8302: xls::dslx::DeduceAndResolve()
0x560aa3fe14d1: xls::dslx::InstantiateParametricArgs()
0x560aa3fe1fb3: xls::dslx::DeduceInvocation()
0x560aa3fb380b: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleInvocation()
0x560aa4355f5b: xls::dslx::Invocation::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3fa8302: xls::dslx::DeduceAndResolve()
0x560aa3fb3925: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
0x560aa435706b: xls::dslx::Let::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3fab334: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x560aa4356eeb: xls::dslx::Statement::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3fae8a8: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x560aa43566fb: xls::dslx::Block::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3f7e988: xls::dslx::TypecheckInvocation()
0x560aa3f77838: std::__1::__function::__func<>::operator()()
0x560aa3fe11be: xls::dslx::DeduceInstantiation()
0x560aa3fd06a7: xls::dslx::(anonymous namespace)::DeduceSpawn()
0x560aa3fb79fb: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleSpawn()
0x560aa435613b: xls::dslx::Spawn::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
0x560aa3fab334: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x560aa4356eeb: xls::dslx::Statement::Accept()
0x560aa3fa7d8f: xls::dslx::Deduce()
0x560aa3f77575: std::__1::__function::__func<>::operator()()
0x560aa406bab8: xls::dslx::DeduceCtx::Deduce()
```

cdleary, 2024-04-02:
@rw1nkler thanks for pointing this out -- can you make a new bug to track?