google/xls #1363: macosx nighly build are failing

proppy, 2024-03-29:
See https://github.com/google/xls/actions/runs/7826040226/job/21351214876
Possibly related to: https://github.com/bazelbuild/bazel/issues/18762
Tentative fix: bump https://github.com/bazel-contrib/toolchains_llvm ?

cdleary, 2024-04-11:
Was playing on my home laptop and bumping it up didn't work, I tried to make the LLVM toolchains optionally register because it seems like things work "just fine" when you don't use the hermetic toolchain. Made #1378 but I'm not sure I got the rule right yet to do the conditional registration.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): The macOS CI and release build workflows in `.github/workflows/` and the Bzlmod toolchain configuration in `MODULE.bazel` have been updated and fixed following the 2024 nightly build breakage. (Closed by Gemini)

closed 2026-11-14 by allight