google/xls #1365: Codegen Error For Pipelined Designs

sazadur, 2024-03-29:
I have just installed the XLS in an apptainer and I was trying to run the examples from here - https://google.github.io/xls/tutorials/xlscc_overview/

The combinational one ran successfully but when i was trying to run the pipelined one, it was throwing some errors. I was able to generate the IR and OPT IR files, but the error keeps popping up during verilog generation.
**To Reproduce**
Steps to reproduce the behavior:
1. Generate the IR and OPT IR from the XLS overview page and then attempt to generate the RTL using codegen

The following error pops up.

**ERROR: Unknown command line flag 'entry'. Did you mean: top ?**

**Screenshots**
[image]

**Environment (this can be helpful for troubleshooting):**
 - OS: [e.g. Ubuntu 22.04]
 - Versions [e.g. 22.04]
 - Apptainer

sazadur, 2024-04-01:
@spurserh , @ericastor - Hi, can you please suggest any possible resolution to this issue?

cdleary, 2024-04-01:
I'm guessing that the XLS[cc] tutorial is out of date and it's called "top" now instead of entry, and the xlscc tutorial needs to be updated? Pass `--help` to any binary to see flags. Does it work when you pass top as it suggests?

sazadur, 2024-04-01:
@cdleary, If I pass top, then a new error pops up. Please note that, I am trying the exact same tutorial suggested in the [XLS page](https://google.github.io/xls/tutorials/xlscc_overview/) for test_unroll module.  

"Error: UNIMPLEMENTED: Data path reset not supported"

cdleary, 2024-04-01:
Sounds like the XLS[cc] docs need to be updated.

Note that the folks who maintain XLS[cc] (the C++ frontend) have different support bandwidth available from the XLS team which builds/maintains the DSLX frontend (which is why xlscc is in the `contrib/` directory).