google/xls #1386: [enhancement] Expose min_delay op in DSLX

hongted, 2024-04-18:
### What's hard to do? (limit 100 words)

Currently IR supports the addition of a MIN_DELAY op to impose scheduling constraint between two nodes.  This should be exposed to DSLX.

https://github.com/google/xls/blob/f231771f048112e74e23f9134a56606c16cdf634/xls/ir/ir_parser_test.cc#L886

### Current best alternative workaround (limit 100 words)

Currently best alternative is the exposed scheduling constraint between send/recv channels.

### Your view of the "best case XLS enhancement" (limit 100 words)

A min_delay op should be added to DSLX.

cdleary, 2024-04-19:
@hongted Are you proposing we introduce the first "cycle count" oriented construct into the DSL, or just that we make a way to sequence tokens with respect to other tokens independent of `after_all`?

hongted, 2024-04-22:
For this, I'm proposing adding a "cycle count" oriented construct into the DSL.  We already have it in the IR, and this issue would be to expose it.