google/xls #1412: Cannot map parametric function

grebe, 2024-05-16:
Mapping a parametric function doesn't work, e.g.

```
fn f<N:u32>(x: u32) -> u32 { N + x }
fn main() { map(u32[4]:[0, 1, 2, 3], f<u32:4>) }
```

hzeller, 2024-06-17:
Also duplicate of https://github.com/google/xls/issues/1063

mikex-oss, 2024-06-17:
I think #1063 is a specific case of this bug for which there can be no available workaround. I.e. you can't simply wrap the parametric instantiation because it depends on parent parameters.

This more generalized issue can be worked around as indicated in https://github.com/google/xls/issues/1063#issuecomment-1633497494.

Not sure what this means for bug management...

proppy, 2024-06-17:
Let's keep this open then, and #1063 as "related" then.

Sorry for the nosie :)

commit 28b4476e (2024-10-21): Support `map(arr, f<ExplicitParametrics>)`.

Previously, parsing would fail at the open angle bracket, and type checking was not equipped to handle this.

Context: https://github.com/google/xls/issues/1412

Fixes #1412

PiperOrigin-RevId: 688228006
closed 2024-10-21 by copybara-service[bot]