google/xls #1425: Cannot use default parameter value if the struct is imported

m-torhan, 2024-05-22:
**Describe the bug**
When a parametrized struct is imported, it is not possible to set default value of its parameter.

**To Reproduce**
Steps to reproduce the behavior:
1. Write following code:

`bar.x`
```rust
pub struct Bar<SIZE:u32 = {u32:6}> {
    a: uN[SIZE],
}
```
`foo.x`
```rust
import bar;

type Bar = bar::Bar;

proc foo {
    ch_r: chan<Bar<u32:4>> in;

    config(ch_r: chan<Bar<u32:4>> in) {(ch_r,)}

    init {}

    next (tok: token, state: ()) {
        let (tok, r) = recv(tok, ch_r);
        trace_fmt!("recv ret {:#x}", r);
    }
}

#[test_proc]
proc foo_test {
    terminator: chan<bool> out;

    ch_s: chan<Bar<u32:4>> out;

    config (terminator: chan<bool> out) {
        let (ch_s, ch_r) = chan<Bar<u32:4>>("bar");
        spawn foo(ch_r);
        (terminator, ch_s)
    }

    init { }

    next (tok:token, state: ()) {
        let bar = Bar<u32:4> { a : u4:0x3 };
        let tok = send(tok, ch_s, bar);
        trace_fmt!("sent {:#x}", bar);

        let tok = send(tok, terminator, true);
    }
}
```
2. Build it with the following bazel targets
```bazel
xls_dslx_library(
    name = "bar_dslx",
    srcs = ["bar.x"],
)

xls_dslx_library(
    name = "foo_dslx",
    srcs = ["foo.x"],
    deps = [":bar_dslx"]
)

xls_dslx_test(
    name = "foo_dslx_test",
    library = ":foo_dslx",
)
```
```sh
bazel build -- //:foo_dslx
```
3. See error
```none
ERROR: /.../xls/BUILD:71:17: Parsing and type checking DSLX source files of target foo_dslx failed: (Exit 255): bash
failed: error executing command (from target //:foo_dslx) /bin/bash -c ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
F0522 12:15:56.082579       3 type_info.h:209] Check failed: key->owner() == module_ (0x64ff12314320 vs. 0x64ff123125b0)
*** Check failure stack trace: ***
    @     0x64fee0e689f4  absl::lts_20240116::log_internal::LogMessage::SendToLog()
    @     0x64fee0e686c8  absl::lts_20240116::log_internal::LogMessage::Flush()
    @     0x64fee0e68d99  absl::lts_20240116::log_internal::LogMessageFatal::~LogMessageFatal()
    @     0x64fedefdca9a  xls::dslx::TypeInfo::SetItem()
    @     0x64fedf04b9e8  xls::dslx::internal::ParametricInstantiator::ParametricInstantiator()
    @     0x64fedf050267  xls::dslx::internal::StructInstantiator::StructInstantiator()
    @     0x64fedf04ffc2  xls::dslx::internal::StructInstantiator::Make()
    @     0x64fedf04b144  xls::dslx::InstantiateStruct()
    @     0x64fedf043541  xls::dslx::DeduceStructInstance()
    @     0x64fedf01aa6b  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStructInstance()
    @     0x64fedf4059ab  xls::dslx::StructInstance::Accept()
    @     0x64fedf00aacf  xls::dslx::Deduce()
    @     0x64fedefd6aa5  std::__1::__function::__func<>::operator()()
    @     0x64fedf0d2858  xls::dslx::DeduceCtx::Deduce()
    @     0x64fedf00b042  xls::dslx::DeduceAndResolve()
    @     0x64fedf016925  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleLet()
    @     0x64fedf4065fb  xls::dslx::Let::Accept()
    @     0x64fedf00aacf  xls::dslx::Deduce()
    @     0x64fedefd6aa5  std::__1::__function::__func<>::operator()()
    @     0x64fedf0d2858  xls::dslx::DeduceCtx::Deduce()
    @     0x64fedf00d9e4  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
    @     0x64fedf40647b  xls::dslx::Statement::Accept()
    @     0x64fedf00aacf  xls::dslx::Deduce()
    @     0x64fedefd6aa5  std::__1::__function::__func<>::operator()()
    @     0x64fedf0d2858  xls::dslx::DeduceCtx::Deduce()
    @     0x64fedf011058  xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
    @     0x64fedf405c8b  xls::dslx::Block::Accept()
    @     0x64fedf00aacf  xls::dslx::Deduce()
    @     0x64fedefd6aa5  std::__1::__function::__func<>::operator()()
    @     0x64fedf0d2858  xls::dslx::DeduceCtx::Deduce()
    @     0x64fedf00b042  xls::dslx::DeduceAndResolve()
    @     0x64fedefdb3c4  xls::dslx::TypecheckFunction()
    @     0x64fedefd473a  xls::dslx::TypecheckModule()
    @     0x64fedef652d7  xls::dslx::TypecheckModule()
    @     0x64fedef647cc  xls::dslx::ParseAndTypecheck()
    @     0x64fedef5a6cb  xls::dslx::ParseAndTest()
    @     0x64fedeefb8f5  main
    @     0x75e627ac8c88  (unknown)
    @     0x75e627ac8d4c  __libc_start_main
    @     0x64fedeefa6a5  _start
/bin/bash: line 2:     3 Aborted                 (core dumped) bazel-out/k8-opt-exec-2B5CBBC6/bin/xls/dslx/interpreter_main $file --compare=none --execute=false --dslx_path=:${PWD}:bazel-out/k8-fastbuild/bin:bazel-out/k8-fastbuild/bin::bazel-out/k8-fastbuild/bin/ --warnings_as_errors=false
Error parsing and type checking DSLX source file: foo.x
Target //:foo_dslx failed to build
```

**Expected behavior**
The build runs with no errors.

**Environment (this can be helpful for troubleshooting):**
 - OS: Arch Linux x86_64
 - Kernel version: 6.9.1-arch1-1
 - XLS hash: `83d661222ba4c4b2a4ece64dd70d3f0061e4b0bf`

**Additional context**
After moving `Bar` definition to `foo.x` the build does not fail.

dplassgit, 2024-07-25:
Simpler repro case:

```
import bar;

type Bar = bar::Bar;
const aBar = Bar<u32:4> { a : u4:0x3 };

fn foo() -> Bar<u32:4> {
  aBar
}

#[test]
fn foo_test() {
  assert_eq(foo(), aBar);
}
```

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/dslx/frontend/parser.cc:2830-2865` and Type System v2 (`xls/dslx/type_system_v2/type_annotation_utils.cc:325-335`) now support instantiating and aliasing imported parametric structs using their default parametric values, as tested in `xls/dslx/type_system/typecheck_module_test.cc:4211-4285`. (Closed by Gemini)

closed 2026-11-14 by allight