google/xls #1432: [enhancement] Tokenless I/O options

ericastor, 2024-05-26:
### What's hard to do? (limit 100 words)

DSLX currently requires users to provide a token every time they want to do an I/O operation (e.g., `send(tok, chan, data)`). This means that if one wants to do an I/O operation with no constraints that it comes after any other particular operation, you need to create and pass a free token (e.g., `send(join(), chan, data)`).

It would be helpful if there was an easy syntax for I/O operations with no ordering constraints other than data dependencies.

### Current best alternative workaround (limit 100 words)

Continue using a free token.

### Your view of the "best case XLS enhancement" (limit 100 words)

Update the syntax of DSLX to allow tokenless I/O operations.

Options include:
- move the token to an *optional* keyword parameter (`send(chan, data)` and `send(chan, data, after=tok)`)
    - advantage: reads very naturally, makes non-data ordering edges the exception rather than the rule
    - downside: moves our syntax further from Rust in this specific circumstance, though this is a very specific context where our semantics are also quite different
- `send()` and `send_token()` (or other names - separate built-ins for tokenless vs. token I/O)
    - downside: proliferation of I/O operations (combinable with `_if` and, for `recv`, `_non_blocking`)
- overloading `send()` based on the number of operands (if no token, treat as unrestricted)
    - downside: moves our syntax further from Rust by introducing function overloading (and focuses the overloading on parameter ordering and number, the less explicit option)

ericastor, 2024-05-26:
Quick note: this issue is not intended to cover any discussion of return values. For this discussion, let's assume `send` will continue returning a token, and `recv` will continue returning a tuple that includes a token.

ericastor, 2024-05-28:
We're currently leaning towards the optional-keyword-parameter choice here, but are open to discussion.

cdleary, 2024-05-30:
I would suggest not introducing a new non-Rust syntax feature (i.e. Rust does not have and doesn't seem like it's like to grow keyword argument syntax). Largely I suggest this because trying to stay closer to Rust when possible is a useful guiding principle that prevents us from having to constantly make more arbitrary decisions on the language syntax, and we already have builtins that leverage superpowers, so doubling down on what we already do (builtins with superpowers) seems more palatable than adding another difference. Mostly trying to reflect the principles / method of decision making when considering how to grow the language when I was working on the frontend with more of my time.

One other note is that inlay support in the language server makes parameter names appear similarly to kwargs, which is nice. E.g. see the gray inlay annotation here from some Rust code, we could add this to the DSLX language server if we wanted the arg intent to be clearer:
[image]

ericastor, 2024-05-30:
I'm certainly open to more proposals!

BTW, it seems like having kwargs is as much a form of superpower for built-ins as any other sort of overloading would be. If we stick to Rust & say "don't ever overload", then I think our choices are narrowed to either:

1. don't provide any special syntax for I/O without a token parameter (users have to write *something* explicit to mean no dependencies), or
2. add even more versions of `send` and `recv`, which I think would bring us up to 2**3=8 named `recv` built-ins.

Of those two, (1) seems preferable to me - remembering the order of allowed suffixes for `recv` is already a burden - but it comes with a pedagogical downside we've discussed before, since you need to think about tokens even when you don't need to express a non-dataflow ordering dependency. If someone has an option (3) that answers all concerns, that would of course be best.

meheff, 2024-05-31:
The motivation for the tokenless syntax has been lessened substantially with the more flexible token semantics that @ericastor landed a little while back. Previously, you had to use a token produced by a previous operation (or the token parameter) which forced the user to nontrivially handle tokens in every send/recv call and could result in inadvertent serialization by cargo culting the `let tok = op(tok, ...);` pattern multiple times. Now, the form `let tok = op(join(), ...)`[^1] is supported where a token is materialized in situ, and nontrivial token handling only needs to be done when explicit ordering is actually desired.

Given the reduced upside of tokenless send/recvs and the potential downside of introducing a new syntax I'm inclined to leave things as they are. I'm not strictly opposed to new syntax and deviating from Rust when the benefit is sufficiently large, but I don't think this clears the bar. Thanks @ericastor for putting the proposal together and enumerating the space of options.

[^1]: Other syntaxes besides `join()` for materializing a token are proposed in #1442.

closed 2024-05-31 by meheff

proppy, 2024-06-03:
>  If someone has an option (3) that answers all concerns, that would of course be best.

Maybe something inspiried by rust futures could work as a more rust-ish solution:
https://docs.rs/futures/latest/futures/executor/fn.block_on.html
```
send(chan, data).block_on(tok);
```
or context like annotations like proposed in https://tmandry.gitlab.io/blog/posts/2021-12-21-context-capabilities/
```
with tok = token() {
  send(chan, data);
}
```
or `as`-like annotation
```
  send(chan, data) after tok;
```

Those sound like rust-ish solution to me that wouldn't break the guidelines mentioned in https://github.com/google/xls/issues/1432#issuecomment-2138599124.