google/xls #1442: [DSLX] Consider alternate syntax for token introduction

ericastor, 2024-05-30:
### What's hard to do? (limit 100 words)

Currently, introducing a new token that represents "no non-data dependency" involves invoking `join()` with no arguments. There have occasionally been arguments that this is confusing.

### Current best alternative workaround (limit 100 words)

Instruct users to continue using `join()` to introduce new tokens.

### Your view of the "best case XLS enhancement" (limit 100 words)

Options include:
- Make tokens optional in most contexts, so users can omit the token rather than introducing a new one.
    - See #1432 for discussion on why this is currently not a favored approach.
- Do nothing, and let `join()` be the standard way to introduce a new token.
    - Example: `let tok = send(join(), chan, msg);`
- Add a new built-in `token()` (name subject to debate) that takes no parameters and creates a new free token.
    - Example: `let tok = send(token(), chan, msg);`
- Extend `zero!` so that `zero!<token>()` returns a free token.
    - Example: `let tok = send(zero!<token>(), chan, msg);`
- Allow tokens to be instantiated like structs, with syntax like `token {}`.
    - Example: `let tok = send(token {}, chan, msg);`
- Add a new use of the token keyword, so that `token` in expression context (with no following parens or colon) means a free token.
    - Example: `let tok = send(token, chan, msg);`

ericastor, 2024-05-30:
@proppy @cdleary @meheff for discussion.

mikex-oss, 2024-05-31:
> Add a new built-in token() (name subject to debate) that takes no parameters and creates a new free token.

This feels like something quite natural in other languages (e.g. Python) where you can call the type to construct it, but it doesn't seem like this is something you do in Rust?

Would another option be something like `token::new()`?

Pro: The intent seems clearer and consistent with Rust [conventions](https://rust-lang.github.io/api-guidelines/interoperability.html?highlight=default#types-eagerly-implement-common-traits-c-common-traits:~:text=new%20is%20the%20constructor%20convention%20in%20Rust%2C%20and%20users%20expect%20it%20to%20exist%2C%20so%20if%20it%20is%20reasonable%20for%20the%20basic%20constructor%20to%20take%20no%20arguments%2C%20then%20it%20should).
Con: Unclear if users would just use `join()` anyway since it's less characters to type.

meheff, 2024-05-31:
I'm not a Rust expert but poking around it seems Rust provides a few forms for constructing unit types depending on how the types are defined:

```
// empty struct
struct Foo {};
let f = Foo {};

// unit struct
struct Foo;
let f = Foo;

// tuple struct
struct Foo();
let f = Foo();
```

You can also define a unit type with an enum but the construction options are basically the same as above but with a `Foo::` qualifier in front of it. You can also define a factory on the type:

```
// struct with factory
struct Foo {}
impl Foo {
  fn new() -> Foo { Foo{} }
}
let f = Foo::new();
```

So if we're following rust, just about all the options are on the table: `token`, `token{}`, `token()`, and `token::new()`. I'm inclined to go with `token` as it's the shortest.

ericastor, 2024-05-31:
Any reason not to go with `join()`?

hzeller, 2024-05-31:
The reason _not_ to go with `join()` that does join nothing would sounds odd and would a reader have to do a double-take.

Having said that, it is _technically_ correct (no preconditions!), one can get used to it (no double take after a while) and it would probably be _more_ confusing if there was a different _make a no-precondition token_ mechanism than what could already be expressed with `join()`.

But if these are the choices we currently have to settle on, then just using `join()` is probably the last problematic.

(FWIW, I still _strongly_ favor an optional token syntax #1432 as it is less confusing and better expresses the intent.)

meheff, 2024-05-31:
I agree with Henner about join being potentially confusing though technically correct (the best kind of correct!) so I'd say we support both `join()` (because not supporting a zero-element join would be surprising and arbitrary) and `token`. True, this is two ways to do the same thing, but I'm not too concerned about this minor violation of orthogonality.

proppy, 2024-06-03:
See below my takes on each of the syntax options (labeled for easier cross referencing):

> A. Make tokens optional in most contexts, so users can omit the token rather than introducing a new one.
> See https://github.com/google/xls/issues/1432 for discussion on why this is currently not a favored approach.

I really like this, as I suspect that most of the time implicit data dependency would be enough and `token` will be more of a escape-hatch to allow developers to go beyond to express more advanced ordering semantics.

> B. Do nothing, and let join() be the standard way to introduce a new token.
> Example: let tok = send(join(), chan, msg);

I think it'd be cool for this to continue to work, but I would recommend against it being advertised as the idiomatic way (even if "technically correct"):
- It would be documented as a special case  (no-data-dependency) of the `join()` semantic (builtins used for expressing data dependency),  while most likely being the most used variant of `join()` (until A. is done).
- Common action usually benefit from being expressed w/ dedicated vocabulary (even if you can express them w/ the negation of a less-common term).

> C. Add a new built-in token() (name subject to debate) that takes no parameters and creates a new free token.
> Example: let tok = send(token(), chan, msg);

That would be my favored option; imo this options is most coherent w/ the current DSLX design choices: 
- where multiple proc related built-ins are already used to express I/O related operations https://google.github.io/xls/dslx_std/#proc-related-builtins-communicating-sequential-processes
- `chan` initialization also used a builtin-fn like syntax.

> D. Extend zero! so that zero!<token>() returns a free token.
> Example: let tok = send(zero!<token>(), chan, msg);

Same as B., I think it'd be cool for this to work (does it already?), but I would recommend against it being advertised as the idiomatic way for similar reasons:
- common action usually benefit from being expressed w/ dedicated vocabulary
- this syntax is the most verbose of the proposed options

> E. Allow tokens to be instantiated as if they were empty structs, with syntax like token {}.
> Example: let tok = send(token {}, chan, msg);

I don't think we we currently have an example of a built-in struct-type in DSLX (only builtins fns), so this would feel a little bit foreign w/ the current look and feel of the language.

> F. Add a new use of the token keyword, so that token in expression context (with no following parens or colon) means a free token.
Example: let tok = send(token, chan, msg);

Similarly I don't think we have an example of a built-in 'global'-like expression, so this would feel a little bit foreign w/ the current look and feel of the language.

meheff, 2024-06-04:
As mentioned above `token`, `token()`, and `token{}` all are possible legal Rust ways to construct an object depending on the way the type is defined. I agree with @proppy that `token()` feels the most natural, so let's go with that.

closed 2024-06-11 by copybara-service[bot]