google/xls #1447: Add --config=asan test in CI

hzeller, 2024-05-31:
To make sure to not accidentally introduce memory leaks or invalid memory access issues, we should run the tests with AddressSanitizer in the CI (maybe as separate after successful 'regular' testing, as it can be quite a bit slower, and we should only run it after we know that the regular test execution passes).

The [.bazelrc already has the configuration](https://github.com/google/xls/blob/d2da1045eb2e55ac55a4bf3f1c8ef974244a2fd7/.bazelrc#L31-L38) prepared for this.

Context: PR #1443 would benefit from an ASAN run.

hzeller, 2024-05-31:
Survey: This is the subset of tests that seem to work currently with asan.

Some are suffering the #1449 compile issue, others are not working in the context of fuzzing, yet others don't work due to the use of cython. The remaining 1358 tests pass.

```
bazel test --config=asan --keep_going -- //xls/... -//xls/fuzzer/... -//xls/codegen/...  \
   -//xls/contrib/ice40:wrap_io_test -//xls/contrib/xlscc/unit_tests:translator_verilog_test \
  -//xls/simulation:module_simulator_codegen_test -//xls/simulation:module_simulator_test \
  -//xls/simulation:module_testbench_test -//xls/simulation:testbench_io_test \
  -//xls/simulation:verilog_simulator_test -//xls/simulation:verilog_test_base_test \
  -//xls/synthesis:synthesis_utils_test -//xls/synthesis:timing_characterization_client_test \
  -//xls/tools:simulate_module_main_test -//xls/contrib/xlscc/unit_tests:fuzz_fixed_10_test  \
  -//xls/contrib/xlscc/unit_tests:fuzz_int_10_test
```

hzeller, 2024-05-31:
Do you think adding a asan build would fit the resources @proppy ?

proppy, 2024-06-13:
@hzeller yep, do we want to run those continuously or nightly?

hzeller, 2024-06-25:
I think on every github PR (but not the regular CI on head) and then once nightly.

proppy, 2024-06-26:
> I think on every github PR (but not the regular CI on head) and then once nightly.

that should be a problem!

proppy, 2024-06-26:
@hzeller I assume the build cache for the two config (w/ and w/o asan) would end up being different?

hzeller, 2024-06-26:
Yes. And possibly asan a lot larger.
mmh, that might bring us over the available cache edge, wouldn't it ?

proppy, 2024-06-26:
I suspect that we could run the asan build without cache?

hzeller, 2024-06-26:
Sounds good. I suspect it might run forever though.
Maybe we can narrow the scope of tha asan test in a PR to only the targets affected (I guess some sort of collaboration between git diff and bazel query). That way we can build relevant targets and maybe bail out if it would be too many targets.

That way we could service most of the PRs, but also not hog down resources too much.

(and then, maybe nightly do a full asan ?)