google/xls #1459: Tuple destructuring syntax for "rest of tuple"

mikex-oss, 2024-06-05:
### What's hard to do? (limit 100 words)

Destructuring a tuple in DSLX where you only want to bind a small subset of elements to variables is overly verbose, since you need to throw away each "don't care" element separately.

### Current best alternative workaround (limit 100 words)

Use `_` to mean "don't bind this element in the tuple to a variable" for each individual element you want to ignore.

For example:

https://github.com/google/xls/blob/bc04b2bb786ad3c9f4914c38a5f58c67f57ada29/xls/examples/dslx_intro/prefix_scan_equality.x#L18

https://github.com/google/xls/blob/a3073f96eea81515d2b735ce5dad5e929596c459/xls/dslx/stdlib/std.x#L208

### Your view of the "best case XLS enhancement" (limit 100 words)

Support Rust `..` syntax to ignore the rest of the tuple (https://doc.rust-lang.org/rust-by-example/flow_control/match/destructuring/destructure_tuple.html). The section describes the application for match patterns, but the same works on the LHS of a `let`. 

So in the last example, this would become:

```rust
let (.., div_result) = for 
```

The `..` can capture one arbitrarily long subsequence within the tuple, e.g. for some `foo` N-tuple with `N >= 2`, you can write:

```
let (bar, .., baz) = foo;
```

cdleary, 2024-06-15:
+1 this would be quite nice to support, and can follow our "feature that matches Rust" principle!

dplassgit, 2024-08-06:
TODO:
1. Parser infrastructure
2. Type deduction
3. Bytecode generation
4. IR conversion
5. TBD

closed 2024-08-16 by copybara-service[bot]