google/xls #1464: ir_opt_ir rule does not format opt_ir_args flags correctly

proppy, 2024-06-06:
**Describe the bug**

`ir_opt_ir` rule doesn't seem to forward `opt_ir_args` command line flags correcty to `opt_main`.

**To Reproduce**

add the following attr to an `ir_opt_ir` target:
```
    opt_ir_args = {"inline_procs": "false"},
```
observe that it gets forwarded as `--inline_procs true`, see:
```
  bazel-out/k8-opt-exec-2B5CBBC6/bin/xls/tools/opt_main bazel-out/k8-fastbuild/bin/xls/examples/passthrough_nested_opt_ir.ir --inline_procs true --output_path bazel-out/k8-fastbuild/bin/xls/examples/passthrough_nested_opt_ir.opt.ir --alsologto bazel-out/k8-fastbuild/bin/xls/examples/passthrough_nested_opt_ir.opt.log)
```
**Expected behavior**

This get forwarded as `--inline_procs=false`.

proppy, 2024-06-06:
Seems like https://github.com/google/xls/blob/536d0ab65d87dbf3f41e3c69f3ad403679e6e0fd/xls/build_rules/xls_ir_rules.bzl#L228-L229 should be using `args_to_string`: https://github.com/google/xls/blob/main/xls/build_rules/xls_common_rules.bzl#L107

allight, 2024-06-06:
nah args_to_string is something we want to move away from (and honestly long term we probably don't want to send opt_ir_args as an untyped dict either.

For now that should be changed to 

```
args.add("--{}={}".format(flag, value))
```

closed 2024-06-06 by copybara-service[bot]