google/xls #1466: Add source location directives in DSLX for trace/error logging

mikex-oss, 2024-06-07:
### What's hard to do? (limit 100 words)

In SystemVerilog (and similar constructs exist in various software languages), we can use `` `__FILE__`` and `` `__LINE__`` to access the current source location, which can be useful for traces and other logging.

The example given in LRM 22.13 is:

```
$display("Internal error: null handle at %s, line %d.",
`__FILE__, `__LINE__);
```

This would dump something like:

```
Internal error: null handle at platforms/deepsea/logic/gfc/rtl/foo.sv, line 7.
```

I don't think DSLX users currently have the ability to express this.

### Current best alternative workaround (limit 100 words)

As I don't think this can be expressed through DSLX currently, the source info will be omitted.

### Your view of the "best case XLS enhancement" (limit 100 words)

Provide similar directives in DSLX that users can use in `trace_fmt!` and possibly elsewhere to highlight where the dump is coming from.

meheff, 2024-07-17:
Would https://github.com/google/xls/issues/1471 which adds line and file to every tract output give you what you need here?

mikex-oss, 2024-07-24:
In the case of trace_fmt, sure. I filed this separately since I think it could still be useful more generally, e.g. for non-trace print or logging. 

It looks like Rust also supports these as macros:

* https://doc.rust-lang.org/std/macro.file.html
* https://doc.rust-lang.org/std/macro.line.html