google/xls #1469: Compiler segfaults while emitting verilog with cryptic error message

joonho3020, 2024-06-11:
**Describe the bug**

When trying to generate Verilog for a fairly complex module written in DSLX the compiler segfaults.

**To Reproduce**
Steps to reproduce the behavior:

- Follow the XLS setup instructions for Ubuntu 2024
- Git clone my [XLS fork](https://github.com/joonho3020/xls/tree/joonho-main)
- Run `bazel build --verbose_failures -- //xls/modules/snappy:snappy_decompressor_verilog`

**Expected behavior**
A clear and concise description of what you expected to happen.

I expect to see the generated verilog

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Environment (this can be helpful for troubleshooting):**
 - OS: [e.g. Ubuntu]
 - Versions [e.g. 22.04]

**Additional context**
Add any other context about the problem here.

hzeller, 2024-06-11:
Can you reduce it to a case that can be run with the main repository not the fork ?

Some *.x file that shows the problem, and the commands you operated on it, e.g.

```python
IR_CONVERTER=bazel-bin/xls/dslx/ir_convert/ir_converter_main
IR_OPT=bazel-bin/xls/tools/opt_main
CODEGEN=bazel-bin/xls/tools/codegen_main
${IR_CONVERTER} /tmp/foo.x | ${IR_OPT} --top=${TOP} - | ${CODEGEN} --module_name=foo --generator=combinational -
```

joonho3020, 2024-06-11:
This is a fairly complicated module that imports `proc`s from other files.
I don't think there is a good way of shrink-wrapping the example as the bug is likely due to the complex structure of this block.
Would it make things better if I PR my branch that is causes these problems?

hzeller, 2024-06-11:
Like in any project is is a good courtesy to minimize the input in question when filing an issue. That way, maintainers can reproduce the problem quickly and focus on solving the problem.
If they have to look through thousands of input lines in a complex project first, that is probably not a good use of their time.

There was probably a point where your project compiled and then it stopped working. Focus on what you changed there, which is probably a good starting point in minimizing the issue.

joonho3020, 2024-06-12:
Came up with the shrink-wrapped version.

```
import std;
import xls.modules.snappy.buffer as buff;
import xls.modules.snappy.common;
import xls.modules.snappy.varint as varint;
import xls.examples.ram;

const BYTE     = common::BYTE;
const BUS_BITS = common::BUS_BITS;
const BUS_BYTES = BUS_BITS >> 3;
const BUFFER_BITS = BUS_BITS * u32:50;

const SNAPPY_MAX_HISTORY_BYTES = common::SNAPPY_MAX_HISTORY_BYTES;
const SRAM_WORD_BITS = BYTE;
const SRAM_WORD_CNT = SNAPPY_MAX_HISTORY_BYTES / BUS_BYTES;
const SRAM_WORD_MASK = BYTE;
const SRAM_RW_BEHAVIOR = ram::SimultaneousReadWriteBehavior::READ_BEFORE_WRITE;
const SRAM_NUM_PARTITIONS = u32:1;
// (SRAM_WORD_MASK + SRAM_WORD_BITS - u32:1) / SRAM_WORD_MASK;
const SRAM_INIT = true;
const SRAM_ADDR_BITS = u32:12;
// std::clog2(SRAM_WORD_CNT);
// type WriteReq = ram::WriteReq<SRAM_ADDR_BITS, SRAM_WORD_BITS, SRAM_NUM_PARTITIONS>;
// type WriteWordReq = ram::WriteWordReq<SRAM_NUM_PARTITIONS, SRAM_ADDR_BITS, SRAM_WORD_BITS>;
// type WriteResp = ram::WriteResp;
// type ReadReq = ram::ReadReq<SRAM_ADDR_BITS, SRAM_NUM_PARTITIONS>;
// type ReadWordReq = ram::ReadWordReq<SRAM_NUM_PARTITIONS, SRAM_ADDR_BITS>;
type ReadResp = ram::ReadResp<SRAM_WORD_BITS>;

type RotBuffer = buff::RotBuffer<BUFFER_BITS>;
type BusBytesBundle = common::BusBytesBundle;
type DataBundle = common::DataBundle<BUS_BITS>;
type uBusBits = uN[BUS_BITS];

type CopyInfo = common::CopyInfo;
type LitInfo = common::LitInfo;
type SnpyCmd = common::SnpyCmd;
type SnpyCmdOrData = common::SnpyCmdOrData<BUS_BITS>;

pub enum CommandExecuterFSM : u2 {
  SET_COMMAND = 0,
  SHIP_LITERALS = 1,
  PERFORM_COPY = 2
}

struct SnappyCommandExecuterState {
  fsm: CommandExecuterFSM,
  cur_cmd: SnpyCmd,
  cmdordata: SnpyCmdOrData,
  historybuffer_ptr: u32, // assume filesizes <= 4GB
  cur_litlen_shipped: u32,
  cur_copy_databundle: DataBundle
}

proc SnappyCommandExecuter {
  cmdordata_in: chan<SnpyCmdOrData> in;
  decompressed_output: chan<DataBundle> out;

  sram_rd_req_s: chan<ram::ReadReq<SRAM_ADDR_BITS, SRAM_NUM_PARTITIONS>>[BUS_BYTES] out;
  sram_rd_resp_r: chan<ram::ReadResp<SRAM_WORD_BITS>>[BUS_BYTES] in;
  sram_wr_req_s: chan<ram::WriteReq<SRAM_ADDR_BITS, SRAM_WORD_BITS, SRAM_NUM_PARTITIONS>>[BUS_BYTES] out;
  sram_wr_resp_r: chan<ram::WriteResp>[BUS_BYTES] in;

  init {
    (zero!<SnappyCommandExecuterState>())
  }

  config(cmdordata_in: chan<SnpyCmdOrData> in,
         decompressed_output: chan<DataBundle> out) {
    let (sram_rd_req_s, sram_rd_req_r) = chan<ram::ReadReq<SRAM_ADDR_BITS, SRAM_NUM_PARTITIONS>>[BUS_BYTES]("sram_rd_req");
    let (sram_rd_resp_s, sram_rd_resp_r) = chan<ram::ReadResp<SRAM_WORD_BITS>>[BUS_BYTES]("sram_rd_resp");
    let (sram_wr_req_s, sram_wr_req_r) = chan<ram::WriteReq<SRAM_ADDR_BITS, SRAM_WORD_BITS, SRAM_NUM_PARTITIONS>>[BUS_BYTES]("sram_wr_req");
    let (sram_wr_resp_s, sram_wr_resp_r) = chan<ram::WriteResp>[BUS_BYTES]("sram_wr_resp");

    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[0], sram_rd_resp_s[0],
        sram_wr_req_r[0], sram_wr_resp_s[0]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[1], sram_rd_resp_s[1],
        sram_wr_req_r[1], sram_wr_resp_s[1]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[2], sram_rd_resp_s[2],
        sram_wr_req_r[2], sram_wr_resp_s[2]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[3], sram_rd_resp_s[3],
        sram_wr_req_r[3], sram_wr_resp_s[3]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[4], sram_rd_resp_s[4],
        sram_wr_req_r[4], sram_wr_resp_s[4]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[5], sram_rd_resp_s[5],
        sram_wr_req_r[5], sram_wr_resp_s[5]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[6], sram_rd_resp_s[6],
        sram_wr_req_r[6], sram_wr_resp_s[6]);
    spawn ram::RamModel<SRAM_WORD_BITS, SRAM_WORD_CNT, SRAM_WORD_MASK,
      SRAM_RW_BEHAVIOR, SRAM_INIT> (
        sram_rd_req_r[7], sram_rd_resp_s[7],
        sram_wr_req_r[7], sram_wr_resp_s[7]);

    (cmdordata_in, decompressed_output,
     sram_rd_req_s, sram_rd_resp_r,
     sram_wr_req_s, sram_wr_resp_r)
  }

  next(tok: token, state: SnappyCommandExecuterState) {
    state
  }
}
```

Fails to execute `${IR_CONVERTER} <above file>.x`.

proppy, 2024-06-13:
@joonho3020 your snippet uses an external file.
```
import xls.modules.snappy.buffer as buff;
```
can you try to use bit.ly/learn-xls to reproduce the error?

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): Although the original reproducer referenced an external fork (`joonho3020/xls`), we should actually check and verify the underlying Verilog codegen crash before closing this issue. (Summarized by Gemini)