google/xls #1470: Compiler bug when generating SRAM blocks using a `for` statement

joonho3020, 2024-06-11:
**Describe the bug**
A clear and concise description of what the bug is.

Compiler errors out when trying to run functional simulation for a fairly complex block.

**To Reproduce**
Steps to reproduce the behavior:

- Follow the XLS setup instructions for Ubuntu 2024
- Git clone my [XLS fork](https://github.com/joonho3020/xls/tree/for-loop-issue)
- Run `bazel run -- //xls/modules/snappy:snappy_decomp_test --alsologtostderr --compare=none`

**Expected behavior**
A clear and concise description of what you expected to happen.

Simulation runs. However I get a compile error here : [error source](https://github.com/joonho3020/xls/blob/1ab579996fabd05a39b35818ce71ad2e56b21f8f/xls/modules/snappy/command_executer.x#L103)

If I hand unroll the above loop, I can run simulations without errors, so I'm assuming that this is a frontend compiler bug.

**Screenshots**
If applicable, add screenshots to help explain your problem.

https://github.com/joonho3020/xls/blob/1ab579996fabd05a39b35818ce71ad2e56b21f8f/xls/modules/snappy/command_executer.x#L103

**Environment (this can be helpful for troubleshooting):**
 - OS: [e.g. Ubuntu]
 - Versions [e.g. 22.04]

**Additional context**
Add any other context about the problem here.

hzeller, 2024-06-11:
Same comment like https://github.com/google/xls/issues/1469#issuecomment-2161463089 applies here.

Can you provide a reproducible example that runs against the head of the main repository not your personal fork ?

joonho3020, 2024-06-11:
Same as #1469 : would it make is easier if I PR the module in the XLS repo?

proppy, 2024-06-13:
@joonho3020 can you try to use https://bit.ly/xls-playground to reproduce the error?

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): Although the original reproducer was on an external fork branch (`joonho3020/xls:for-loop-issue`) under legacy TIv1, we should actually check and verify the SRAM `for` loop behavior under `xls/dslx/type_system_v2/` before closing this issue. (Summarized by Gemini)