google/xls #1472: interpreter failure w/ const_expr bit slice

proppy, 2024-06-12:
**Describe the bug**

The following snippets:
```
import float32;

pub fn double_fraction_carry(f: float32::F32) -> (uN[float32::F32_FRACTION_SZ], u1) {
    let f = f.fraction as uN[float32::F32_FRACTION_SZ + u32:1];
    let f_x2 = f + f;
    (f_x2[0+:float32::F32_FRACTION_SZ], f_x2[float32::F32_FRACTION_SZ+:u1])
}
```

Produce the following error:
```
E0612 09:13:18.051282    2505 deduce.cc:240] INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/deduce.cc:240) type->IsMeta() uN[32]
0x5978de2d938a: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x5978dc59b5b9: absl::lts_20240116::StatusOr<>::StatusOr<>()
0x5978dc60bbfa: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleTypeRef()
0x5978dca0e83b: xls::dslx::TypeRef::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc60cea1: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleTypeRefTypeAnnotation()
0x5978dca0ecfb: xls::dslx::TypeRefTypeAnnotation::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc626a44: xls::dslx::(anonymous namespace)::DeduceSliceType()
0x5978dc6135cd: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleIndex()
0x5978dca0dcab: xls::dslx::Index::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc635690: xls::dslx::DeduceXlsTuple()
0x5978dc619a6b: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleXlsTuple()
0x5978dca0debb: xls::dslx::XlsTuple::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc60b424: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleStatement()
0x5978dca1043b: xls::dslx::Statement::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc60ea98: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleBlock()
0x5978dca0fc4b: xls::dslx::Block::Accept()
0x5978dc60850f: xls::dslx::Deduce()
0x5978dc5d4485: std::__1::__function::__func<>::operator()()
0x5978dc6d02a8: xls::dslx::DeduceCtx::Deduce()
0x5978dc608a82: xls::dslx::DeduceAndResolve()
0x5978dc5d8da4: xls::dslx::TypecheckFunction()
0x5978dc5d1cdb: xls::dslx::TypecheckModule()
0x5978dc5602d7: xls::dslx::TypecheckModule()
0x5978dc55f7c7: xls::dslx::ParseAndTypecheck()
0x5978dc5551ea: xls::dslx::ParseAndTest()
0x5978dc4f63b5: main
0x7a6b41306d90: [unknown]
```

mikex-oss, 2024-06-12:
Shouldn't `f_x2[0+:float32::F32_FRACTION_SZ]` be just `f_x2[0:float32::F32_FRACTION_SZ]`? Since `F32_FRACTION_SZ` is just a constant bit index and not a bit type.

proppy, 2024-06-12:
> f_x2[0+:float32::F32_FRACTION_SZ]

you're right! but we should still error out w/ an actionable message rather than a traceback.

closed 2024-06-18 by copybara-service[bot]