google/xls #1474: add_with_carry function not found

proppy, 2024-06-12:
**Describe the bug**

Trying to use the documented built-in `add_with_carry` like in the snippet below:
```
import float32;
import std;

pub fn double_fraction_carry(f: float32::F32) -> (uN[float32::F32_FRACTION_SZ], u1) {
    let f = f.fraction as uN[u32:23 + u32:1];
    let (c, f_x2) = add_with_carry<24>(f, f);
    (f_x2[0+:u23], c)
}
```

fails with the following error:
```
xls_work_dir/user_module.x:8:35-8:41
0006: pub fn double_fraction_carry(f: float32::F32) -> (uN[float32::F32_FRACTION_SZ], u1) {
0007:     let f = f.fraction as uN[u32:23 + u32:1];
0008:     let (c, f_x2) = add_with_carry(f, f);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^----^ IrConversionError: Could not find name for invocation: `add_with_carry`; available: [double_fraction_carry]
0009:     (f_x2[0+:u23], c)
0010: }

```

cdleary, 2024-06-15:
If we're not going to add an IR node for it (and I don't think we have any active plans to) I think we should probably remove this in favor of stdlib userspace helper function.

cdleary, 2024-06-15:
(And note this is technically a frontend issue since it's an error in IR conversion.)

dplassgit, 2024-07-30:
> If we're not going to add an IR node for it (and I don't think we have any active plans to) I think we should probably remove this in favor of stdlib userspace helper function.

This seems to be the consensus. @proppy do you have a preferred format/template/already existing FR for the stdlib?

closed 2024-08-01 by copybara-service[bot]

proppy, 2024-09-02:
> we'll remove it from the FE.

Did we also update the doc?

> @proppy do you have a preferred format/template/already existing FR for the stdlib?

@dplassgit https://github.com/google/xls/issues/new?assignees=&labels=enhancement&projects=&template=enhancement-proposal.yml&title=%5Benhancement%5D+ would do also filed # to discuss how to consolidate builtins and stdlib namespace #1579

commit 35ccaa58 (2024-08-01): Remove the built-in `add_with_carry`, as it was never fully implemented, and probably would be better as a `std` library function.

https://github.com/google/xls/issues/1474

Fixes #1474

PiperOrigin-RevId: 658383561