google/xls #1476: DSLX formatter removes comments above proc channel declarations

mikex-oss, 2024-06-12:
**Describe the bug**
After running through dslx autofmt, the comments describing a channel declaration are gone.

**To Reproduce**
Steps to reproduce the behavior:
1. Take the example [adder proc](https://google.github.io/xls/tutorials/what_is_a_proc/#channels:~:text=to%20a%20third%3A-,pub%20proc%20adder%20%7B,-A%3A%20chan) in the proc tutorial and add comments above each channel declaration.
```
pub proc adder {
  // foo
  A: chan<u32> in;
  // bar
  B: chan<u32> in;
  // baz
  C: chan<u32> out;
```

2. Run the formatter or dslx_fmt_test.
3. You will see the comments removed; test diff shown below:

```
<   // foo
<   A: chan<u32> in;
<   // bar
<   B: chan<u32> in;
<   // baz
<   C: chan<u32> out;
---
>     A: chan<u32> in;
>     B: chan<u32> in;
>     C: chan<u32> out;
```

**Expected behavior**
Comments should be preserved.

mikex-oss, 2024-06-12:
Note inline comments and blank lines (between comment and declaration, or between declarations) doesn't seem to workaround this issue.

cdleary, 2024-08-18:
We think this should be fixed by #1535

closed 2024-08-18 by cdleary