google/xls #1477: [crash] Can not use `map()` on a function that uses `assert!()` or `fail!()`

hzeller, 2024-06-13:
**Describe the bug**

Consider the following simple use of the `map()` function, calling a function that uses an `assert!()`
### Input
Let's consider this `/tmp/foo.x`
```rust

fn mapped_fun(x: u32) -> u32 {
    assert!(x != u32:0xffff_ffff, "overflow");
    x + u32:1
}

pub fn foo(x: u32[8]) -> u32[8] {
    map(x, mapped_fun)
}
```

This fails to compile, as the use of `assert` means that somewhere in the processing, a `__token` and `__activated` are added to the parameter list, so now the map operation that expects to call a function with exactly one parameter fails:

```python
bazel build -c opt xls/dslx/ir_convert:ir_converter_main
bazel-bin/xls/dslx/ir_convert/ir_converter_main /tmp/foo.x
```

### Result
```
E0612 16:56:41.029630  923384 verify_node.cc:755] INTERNAL: XLS_RET_CHECK failure (xls/ir/verify_node.cc:755) 1 == map->to_apply()->params().size() (1 vs. 3) 
0x5635984e2a2a: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x5635984b493c: xls::(anonymous namespace)::NodeChecker::HandleMap()
0x56359848b78f: xls::Node::VisitSingleNode()
0x5635984af3da: xls::VerifyNode()
0x5635984410eb: xls::BuilderBase::CreateBValue()
0x5635984455a8: xls::BuilderBase::AddNode<>()
0x563598445503: xls::BuilderBase::Map()
0x563598208488: std::__1::__function::__func<>::operator()()
0x56359820467f: std::__1::__function::__func<>::operator()()
0x5635981c7d1d: xls::dslx::FunctionConverter::DefWithStatus()
0x5635981c8754: xls::dslx::FunctionConverter::Def()
0x5635981daea4: xls::dslx::FunctionConverter::HandleMap()
0x5635981e09fc: xls::dslx::FunctionConverter::HandleInvocation()
0x5635981fa712: xls::dslx::FunctionConverterVisitor::HandleInvocation()
0x5635984336bb: xls::dslx::Invocation::Accept()
```

dank-openai, 2025-02-28:
This bug has bitten me too.

allight, 2025-04-18:
Me too. I wonder if it might be worth it to just desugar map to the for-loop during irconvert then all this gets handled.

We can delete the map op from the ir too if we do this.

allight, 2025-04-18:
I've actually got a fix for this in review now.

closed 2025-04-21 by copybara-service[bot]