google/xls #1479: Map return type inference not working

mikex-oss, 2024-06-13:
**Describe the bug**
Cannot directly return a `map` as a function return value, due to type mismatch. It seems to infer an empty tuple. Instead, you must assign the map result to a variable and return that.

**To Reproduce**
Steps to reproduce the behavior:
1. Take the simple program in the `map` documentation:
```
import std;

fn main(x: s3[3]) -> s3[3] {
  let y: s3[3] = map(x, std::abs);
  y
}
```
2. Reduce this to a direct return of the `map`:
```
import std;

fn main(x: s3[3]) -> s3[3] {
  map(x, std::abs);
}
```
3. Build the dslx_library.
4. Get a build error:
```shell
0003:   fn main(x: s3[3]) -> s3[3] {
       ____________________________^
0004: |   map(x, std::abs);
0005: | }
      |_^ XlsTypeError: Return type of function body for 'main' did not match the annotated return type.
Type mismatch:
   ()
vs sN[3][3]
```

**Expected behavior**
Snippets in Steps 1 and 2 should be equivalent and both compile.

hzeller, 2024-06-13:
you have a semicolon at the end of your return, is that the issue ? Because now you're implicitly returning `()`

mikex-oss, 2024-06-13:
@ericastor pointed out that this is WAI, because of the trailing semi-colon. This causes the next line to be used as the return value, and since there is no next line in the function, it returns the empty tuple.

This behavior also exists in Rust.

mikex-oss, 2024-06-13:
@hzeller yes correct.

closed 2024-06-13 by mikex-oss