google/xls #1493: [enhancement] [DSLX] `priority_sel` builtin should support a specified default value

ericastor, 2024-06-18:
### What's hard to do? (limit 100 words)

Currently, priority-select always uses a zero default value, restricting its use to Bits-typed inputs.

### Current best alternative workaround (limit 100 words)

Even for Bits-typed inputs, using a nonzero default value requires adding another bit to the selector & manually setting it to the NOR of all the other bits, and adding a final case - and the result still puts a zero default-value in the circuit emitted unless codegen happens to prove that the selector is never zero.

For non-Bits-typed inputs, the only option is to use a `select` and map your selector to integer values.

### Your view of the "best case XLS enhancement" (limit 100 words)

`priority_sel` should allow the user to specify a default value, or a version that does should be added.

dplassgit, 2024-07-26:
The IR already supports a default but dslx doesn't expose it yet.

grebe, 2024-07-29:
I support updating the `priority_sel()` dslx builtin to match IR in having a default, although now that `match` converts to priority select by default now, perhaps the builtin is superfluous?

cdleary, 2024-07-30:
Note that the builtin form in particular is useful for fuzzing, don't need to go through trying to make different match constructs (and how match constructs happen to IR convert) to fuzz its args directly.

commit 69f3b7cd (2024-07-31): Change the dslx `priority_sel` built-in to accept a third argument, which is the default return value.

Previously it used 0 (of the appropriate size) for the default return value if the selector's bit pattern didn't correspond to any of the provided cases.

https://github.com/google/xls/issues/1493

Fixes #1493

PiperOrigin-RevId: 658020003
closed 2024-07-31 by copybara-service[bot]