google/xls #1498: ParseAndTypecheck() errors incorrectly for concatenated enum values.

richmckeever, 2024-06-24:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
ParseAndTypecheck() with the following DSLX:

```
#![allow(nonstandard_constant_naming)]
#![allow(nonstandard_member_naming)]

#[sv_type("a::enum_t")]
pub enum enum_t : bits[2] {
    kElem0 = 0,
    kElem1 = 1,
}

pub const p = enum_t::kElem0 ++ enum_t::kElem1;
```

errors with: "Concatenation requires operand types to be either both-arrays or both-bits."

**Expected behavior**
This should not error, since both enum elements are actually bits.

cdleary, 2024-06-24:
This was intentional, the user has to cast to the underlying bits type (and I think unsigned, I believe signed doesn't allow concat) in order to concatenate (note in Rust there's not a bit concatenation but enums are not generally weakly typed). The error message should be better though!

meheff, 2024-06-25:
Fixed by 7d28a48116206e3915150adb156cb0041a2519c2

closed 2024-06-25 by meheff