google/xls #1515: Internal Error: Proc ID was not found in arg mapping

bgelb-openai, 2024-07-17:
**Describe the bug**
Getting this internal error, and I am not sure what it means.

E0717 17:47:56.946600   15000 command_line_utils.cc:45] Could not extract a textual position from error message: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.
Error: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.

**To Reproduce**
I am using the xls playground colab environment.

I defined two procs, tx_proc and rx_proc (neither instances the other).

I am using this dslx invocation:
%%dslx --top=tx_proc --pipeline_stages=1 --reset=reset

I have a proc_test instantiating both procs and connecting them via a channel.

I get this error when attempting to run and I don't understand what the issue is. Per @cdleary internal errors are not expected so it is worth filing an issue.

**Expected behavior**
Need error message to be more actionable so that I can correct the issue.

**Additional context**
Not able to share the code as-is, but can provide more info as needed. I am not sure what is most relevant based on the error message.

proppy, 2024-07-17:
> I am using the xls playground colab environment.

can you update the version of XLS to https://github.com/google/xls/releases/tag/v0.0.0-5481-g069029c49 in the first cell and see if the error is still happening?

bgelb-openai, 2024-07-17:
I updated the version as requested. See same error:

E0717 20:54:30.594561   62822 command_line_utils.cc:46] Could not extract a textual position from error message: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.
Error: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.

proppy, 2024-07-17:
@bgelb-openai  are you able to reproduce the error w/ a minimal `%%dslx` cell that you can share?

bgelb-openai, 2024-07-17:
I was not expecting to be able to do this easily, but looks like the answer is yes:

```
%%dslx --top=tx_proc --pipeline_stages=1 --reset=reset

proc tx_proc {
    packet_in: chan<u32> in;
    packet_out: chan<u32> out;

    init {
        ()
    }

    config(packet_in: chan<u32> in, packet_out: chan<u32> out) {
            (packet_in, packet_out)
    }

    next(state: ()) {
        let (tok, in_pkt, got_pkt) = recv_non_blocking(join(), packet_in, u32:0);
        send_if(tok, packet_out, got_pkt, in_pkt);
    }
}

proc rx_proc {
    packet_in: chan<u32> in;
    packet_out: chan<u32> out;

    init {
        ()
    }

    config(packet_in: chan<u32> in, packet_out: chan<u32> out) {
            (packet_in, packet_out)
    }

    next(state: ()) {
        let (tok, in_pkt, got_pkt) = recv_non_blocking(join(), packet_in, u32:0);
        send_if(tok, packet_out, got_pkt, in_pkt);
    }
}

#[test_proc]
proc simple_test {
    packet_in_s: chan<u32> out;
    packet_out_r: chan<u32> in;

    terminator: chan<bool> out;

    init { () }

    config(terminator: chan<bool> out) {
        let (packet_in_s, packet_in_r) = chan<u32>("input_chan");
        let (packet_out_s, packet_out_r) = chan<u32>("output_chan");
        let (rpacket_s, rpacket_r) = chan<u32>("inbetween_chan");

        spawn tx_proc(packet_in_r, rpacket_s);
        spawn rx_proc(rpacket_r, packet_out_s);
        (packet_in_s, packet_out_r, terminator)
    }

    next(state: ()) {
        let test_pkt = u32:42;
        let tok = send(join(), packet_in_s, test_pkt);
        let (tok, result) = recv(tok, packet_out_r);
        assert_eq(test_pkt, result);

        let tok = send(tok, terminator, true);
    }
}
```

Error:
```
E0717 22:14:00.125137   81841 command_line_utils.cc:46] Could not extract a textual position from error message: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.
Error: INTERNAL: Proc ID "rx_proc:0" was not found in arg mapping.
```

proppy, 2024-07-17:
Sound similar to #1415 ?

Not really possible w/ the playground, but it should work in separate file using the regular command line tools.

bgelb-openai, 2024-07-17:
I added a FAKE_PARAMETER to one of the procs and indeed the error went away! So confirming it is a dupe of 1415.

closed 2024-07-17 by proppy