google/xls #1518: invalid codegen output in with --system_verilog=true

bgelb-openai, 2024-07-25:
**Describe the bug**
I took some dslx code and fed it thorough convert->opt->codegen

There were two different problems that made the SV unparseable.

First, this initial value assignment is not syntactically correct:
```
  wire [31:0] ____state_5_init[16];
  assign ____state_5_init = '{{8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}, {8'h00, 8'h00, 8'h00, 8'h00}};
```

There seems to be an errant ' character at the beginning of the RHS error. When I deleted it by hand, this line parses correctly.

Second, the mapping of the assertions in my dslx code came out mangled:

```
      __llr_tx__packet_out_valid_reg <= llr_tx__packet_out_valid_load_en ? try_tx : __llr_tx__packet_out_valid_reg;
    end
  end
/* <-- had to comment out this section to make it parseable
  can't push onto full queue!: assert property (@(posedge clk) disable iff ($sampled(rst)) ~p0_stage_done | ~(got_new_pkt & ~try_rx)) else $fatal(0, "Assertion failure via assert! @ llr_tx.x:38:12-38:60");
  __llr_tx__llr_tx_0_next_14___itok__llr_tx__q_ack_to___itok__llr_tx__q_ack_to_6___itok__llr_tx__q_ackN_N larger than items!: assert property (@(posedge clk) disable iff ($sampled(rst)) ~p0_stage_done | ~(~timeout & __llr_tx__returning_ack_valid_reg & add_1586 > concat_1587)) else $fatal(0, "Assertion failure via assert! @ llr_tx.x:57:12-57:65");
  __llr_tx__llr_tx_0_next_14___itok__llr_tx__q_ack_to___itok__llr_tx__q_ack_to_7___itok__llr_tx__q_rewindN_N larger than items!: assert property (@(posedge clk) disable iff ($sampled(rst)) ~p0_stage_done | ~(nor_1596 & sub_1569 > concat_1587)) else $fatal(0, "Assertion failure via assert! @ llr_tx.x:47:12-47:65");
  __llr_tx__llr_tx_0_next_16___itok__llr_tx__q_ack_to___itok__llr_tx__q_ack_to_6___itok__llr_tx__q_ackN_N larger than items!: assert property (@(posedge clk) disable iff ($sampled(rst)) ~p0_stage_done | ~(timeout & sub_1588 > concat_1587)) else $fatal(0, "Assertion failure via assert! @ llr_tx.x:57:12-57:65");
  __llr_tx__llr_tx_0_next_16___itok__llr_tx__q_ack_to___itok__llr_tx__q_ack_to_7___itok__llr_tx__q_rewindN_N larger than items!: assert property (@(posedge clk) disable iff ($sampled(rst)) ~p0_stage_done | ~(timeout & sub_1562 > concat_1587)) else $fatal(0, "Assertion failure via assert! @ llr_tx.x:47:12-47:65");
*/
  assign llr_tx__packet_out = __llr_tx__packet_out_reg;
  assign llr_tx__packet_out_vld = __llr_tx__packet_out_valid_reg;
```

There seems to be some text missing in the generated SV at the beginning of the line (at least).

For reference here was the dslx formulation of the assert:

```
    assert!(!q_full(state), "can't push onto full queue!");
```

**Workaround**

I added the flag --use_system_verilog=false and then both issues were resolved.

The state_init assign turned into this:
```
  assign ____state_5_init[0] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[1] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[2] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[3] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[4] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[5] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[6] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[7] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[8] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[9] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[10] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[11] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[12] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[13] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[14] = {8'h00, 8'h00, 8'h00, 8'h00};
  assign ____state_5_init[15] = {8'h00, 8'h00, 8'h00, 8'h00};
```

and no asserts were generated in non-sv mode (thus avoiding the issue entirely).

**Expected behavior**
systemverilog output is syntactically correct and can be parsed by an SV-enabled simulator.

**Environment (this can be helpful for troubleshooting):**
Ubuntu 22.04
XLS release: xls-v0.0.0-5500-g9e21b5623-linux-x64.tar.gz

**Other context**
I can't easily share code. I might be able to try to create a minimal repro, but wanted to check if this rings any bells first.

ericastor, 2024-07-25:
Can you say what simulator is complaining? It might help us understand the problem.

bgelb-openai, 2024-07-25:
simulator is icarus verilog, which has pretty lousy error messages, but here it is.

For the codegen'd assertions:
```
llr_tx.sv:275: syntax error
llr_tx.sv:275: error: Invalid module instantiation
llr_tx.sv:276: syntax error
llr_tx.sv:276: error: Invalid module instantiation
llr_tx.sv:277: syntax error
llr_tx.sv:277: error: Invalid module instantiation
llr_tx.sv:278: syntax error
llr_tx.sv:278: error: Invalid module instantiation
llr_tx.sv:279: syntax error
llr_tx.sv:279: error: Invalid module instantiation
```

Just eyeballing it you can just clearly see its mangled. So I really don't think the error message is telling anything we don't already know. Part of the assertion code is just missing in the codegen output (it starts w/ the string of the error message and its not even quoted).

poking at the array initialization a bit more (the ' character), I see that is actually the proper way to initialize an SV-style unpacked array (my bad, I have never used SV packed arrays in synthesis). I think the issue there is that icarus seems to struggle (or maybe its completely unsupported) to support SV-style unpacked arrays, e.g.:

SV-style:
reg [1:0] x [4];
vs "classic" style:
reg [1:0] x [0:3]; <-- this seems to work as this is what you get in --system_verilog=false mode

So probably can chalk that up to an icarus problem. (aside: is there any option to cause codegen to use unpacked multi-dim arrays in SV mode? icarus aside, unpacked always felt more natural for synthesizable code even though most tools will support both)

bgelb-openai, 2024-07-25:
OK I was able to make a simple repro case:

```
fn test (a: u32, b: u32) -> bool {
  assert!(a == b, "they must be equal");
  a == b
}
```

output sv:

```
module __test__test(
  input wire clk,
  input wire [31:0] a,
  input wire [31:0] b,
  output wire out
);
  // ===== Pipe stage 0:

  // Registers for pipe stage 0:
  reg [31:0] p0_a;
  reg [31:0] p0_b;
  always_ff @ (posedge clk) begin
    p0_a <= a;
    p0_b <= b;
  end

  // ===== Pipe stage 1:
  wire p1_eq_38_comb;
  assign p1_eq_38_comb = p0_a == p0_b;

  // Registers for pipe stage 1:
  reg p1_eq_38;
  always_ff @ (posedge clk) begin
    p1_eq_38 <= p1_eq_38_comb;
  end
  they must be equal: assert property (@(posedge clk) disable iff ($sampled($isunknown(p1_eq_38_comb))) p1_eq_38_comb) else $fatal(0, "Assertion failure via assert! @ test.x:2:10-2:40");
  assign out = p1_eq_38;
endmodule
```

codegen args:
```
codegen_main --generator=pipeline --clock_period_ps=2000 --delay_model=asap7 --pipeline_stages=1 test.opt.ir --output_verilog_path=test.sv --use_system_verilog=true
```

bgelb-openai, 2024-07-25:
retitled the issue to limit it to the assertion codegen problem, as the other issue I don't think its XLS' fault.

mikex-oss, 2024-07-25:
> Part of the assertion code is just missing in the codegen output (it starts w/ the string of the error message and its not even quoted).

It looks like XLS is just incorrectly using the message as the block identifier of the concurrent assertion, not that there is anything missing. `can't push onto full queue!` and `they must be equal` aren't valid block identifiers, which can't have whitespace or an exclamation point.

ericastor, 2024-07-25:
I believe DSLX doesn't currently support assert messages. It expects that the second parameter to `assert!` is a *label*, which is expected to be Verilog-compatible.

bgelb-openai, 2024-07-25:
Thanks for explaining that.  Would it make sense to add some sanity checking of the input in the dslx interpreter? All my unit tests pass in the interpreter so I wasn't really contemplating that it could generate invalid verilog.

meheff, 2024-07-25:
@bgelb-openai Are you sure the '{ assignment is not allowed in SV? Commericial simulators do not seem to have a problem with it. Here's an example (among several) from the LRM (7.12.3):

```
logic [7:0] m [2][2] = '{ '{5, 10}, '{15, 20} };
```

We recently had an issue with verilator not handling the declaration and assignment on the same line when '{ is in play, but that has been fixed by splitting the declaration and assignment into different lines.

Iverilog does not support all of our SystemVerillog constructs. In particular it doesn't support our asserts (and maybe not '{ ?). Switching to emitting verilog solves that problem but then no asserts are emitted I believe.

Agreed about checking that the assert "message" is a valid verilog label and improving the error message.

bgelb-openai, 2024-07-25:
I was confused about ' (i hadn't seen it before) but agree I think its allowed (sorry if my comments above were not clear). So I think there's nothing wrong there other than iverilog limitations.

mikex-oss, 2024-07-25:
I have a change out to add some validation of the assertion label in codegen.

bgelb-openai, 2024-07-25:
is it also a requirement that labels are unique?

closed 2024-07-30 by copybara-service[bot]