google/xls #1527: [enhancement] DSLX tests should be able to set/view proc state

hongted, 2024-08-06:
### What's hard to do? (limit 100 words)

In writing DSLX tests, it is desirable to

Inspect the state of an individual proc so that architectural state can be observed

Inject/set the proc state so that the initial state can be set without going through a long input sequence to get to the desired state.

### Current best alternative workaround (limit 100 words)

Current workaround is to write the test in C++ which has access to the state

### Your view of the "best case XLS enhancement" (limit 100 words)

Provide functionality akin to:

```
proc a_test {
  config ( ...)  {
    let X = spawn A();
    let Y = spawn B();
  }

  next () {
    // < run test>
    assert_eq(X.state.val0, Y.state.val1);
  }
}
```

proppy, 2024-08-19:
related: #843 #906

dplassgit, 2025-05-07:
> related: [#843](https://github.com/google/xls/issues/843) [#906](https://github.com/google/xls/issues/906)

@proppy how are #843 and this issue different?

ericastor, 2025-05-07:
Clarifying question - how does the test proc sync its viewpoint to the proc-under-test to check its state?

The state of a proc is well-defined per proc activation, but if the proc is not idle, its state is not well-defined to any external viewer; different state elements can be updated at different times.

To support this well, do we also need to have a way to express that a test proc step should wait until the proc-under-test is idle?