google/xls #1534: Clean up uids and names in XLS IR

hongted, 2024-08-08:
In the textual IR the identifier of an instruction is given as a string followed by an optional unique numeric id like so:
```
  foo.42: bits[123] = add(x, y)
```
Also legal to omit the id like so:
```
  foo: bits[123] = add(x, y)
```
In the latter case a unique numeric id is assigned to the node at construction time. In both cases the name ("foo") is dropped on the floor and is not written into the IR. The name used when writing out the ir is generated from the opcode and the numeric id ("add.42"). This causes a few problems:

1. deserializing then serializing is not stable because the name is dropped on parsing.
2. no way to give nodes a meaningful name which can help with IR/verilog readability
3. handling of numeric ids is inconsistent with the way we handle other Node attributes.

A better way would be to have the uid given as an (optional) attribute of the node like so:

  foo: bits[123] = add(x, y, uid=42)

And then allowing the string identifier be arbitrary and saved within the Node object itself.

A separate but related issue is how to represent numeric ids for parameter nodes in the textual ir. This is a problem because we do not emit an instruction for parameters in the IR text.  One option is to actually emit the param as a separate instruction when emitting the IR:

```
fn main(p: bits[8], q: bits[8]) -> bits[8] {
  p: bits[8] = param(uid=1)
  q: bits[8] = param(uid=2)
  ret foo: bits[8] = add(p, q, uid=42)
}
```

Or maybe include the uid in the signature like so:
```
  fn main(p: bits[8] (uid=1), q: bits[8] (uid=2)) -> bits[8] {
```
Or something like:
```
  fn main(p#1: bits[8], q#1: bits[8]) -> bits[8] {
```
Changing this will result in some churn in test cases, but I think some script magic may be able to do 95% of the necessary changes by walking all of our source files and transforming only the raw c++ strings (strings of the form R”(...)”)

hongted, 2024-08-08:
Note that ids are now an optional attribute of the node:

ex.
```
  foo: bits[123] = add(x, y, id=42).
```

however the uniqueness issue remainds.  Mixing nodes with explicit ids and nodes without explicit ids (like params), can cause collisions in IDs.  This may result in a IR verification issue or downstream issues.

meheff, 2024-09-07:
I propose adding the id as an optional keyword to the parameter in the parameter list, something like:

```
fn main(p: bits[8] id=42, q: bits[8] id=43) -> bits[8] {
 ...
```

This easier to implement than adding a separate param node in the body and is less fragile as there is no need to keep the parameter name and type in two places.

closed 2024-09-13 by copybara-service[bot]