google/xls #1539: yosys report syntax error for the array expression codegen_main generates

Aaronchan-sz, 2024-08-14:
**Describe the bug**
Yosys reports:
```
1. Executing Verilog-2005 frontend: test.v
test.v:5: ERROR: syntax error, unexpected invalid token
```
This is test.v:
```
module __test__test_main(
  input wire [1:0] idx,
  output wire [2:0] out
);
  wire [2:0] ma[4] = '{3'h1, 3'h2, 3'h3, 3'h4};

  assign out = ma[idx];
endmodule
```

**To Reproduce**
Steps to reproduce the behavior:
test.x source DSLX file:
```
fn test_main(idx:u2)-> u3  {
    let ma:u3[4] = [u3:1, u3:2, u3:3, u3:4];

    ma[idx]
}
```
after ir_converter_main, test.ir
```
package __modul_xls

file_number 0 "__modul_xls.x.tmp"

top fn test(index: bits[2])-> bits[3] {
  literal.1234: bits[3][4] = literal(value=[0, 1, 2, 4], id=1234)
  
  ret val.2234:bits[3] =  literal.1234[index]
}
```
Afte opt_main, test.opt.ir
```
package test

file_number 0 "test.x"

top fn __test__test_main(idx: bits[2]) -> bits[3] {
  ma: bits[3][4] = literal(value=[1, 2, 3, 4], id=8, pos=[(0,1,19)])
  ret array_index.7: bits[3] = array_index(ma, indices=[idx], id=7, pos=[(0,3,6)])
}
```

mikex-oss, 2024-08-14:
To repro, you'll need to provide the codegen options, but it would appear from the array assignment pattern that you are asking XLS to generate SystemVerilog. 

Your Yosys error says "Executing Verilog-2005 frontend" which wouldn't support this type of assignment.

Aaronchan-sz, 2024-08-14:
> To repro, you'll need to provide the codegen options, but it would appear from the array assignment pattern that you are asking XLS to generate SystemVerilog.
> 
> Your Yosys error says "Executing Verilog-2005 frontend" which wouldn't support this type of assignment.

Updated.

mikex-oss, 2024-08-14:
SystemVerilog is the default. You need to turn if off if you're going to use the generated code with a Verilog-2005 compiler.

https://google.github.io/xls/codegen_options/#codegen-mapping

> --use_system_verilog sets if the output should use SystemVerilog constructs such as SystemVerilog array assignments, @always_comb, @always_ff, asserts, covers, etc. True by default.

closed 2024-08-15 by Aaronchan-sz