google/xls #1542: Support anonymous function in map

mikex-oss, 2024-08-15:
### What's hard to do? (limit 100 words)

I've found several times where i want to `map` an array with some simple pattern on each element but don't want to make a top-level function just to do so. It would be nice to support some lambda syntax in maps. 

For example, consider:

```
import std;

struct Foo { bar: u32, baz: u32 }

fn main() {
    let foos: Foo[u32:3] = [Foo { bar: 1, baz: 2 }, Foo { bar: 3, baz: 4 }, Foo { bar: 5, baz: 6 }];
    // want these to be simple maps
    let bars: u32[u32:3] = [foos[0].bar, foos[1].bar, foos[2].bar];
    let sums: u33[u32:3] = [uadd(foos[0].bar, foos[0].baz), uadd(foos[1].bar, foos[1].baz), uadd(foos[2].bar, foos[2].baz)];
}
```

### Current best alternative workaround (limit 100 words)

This can of course be done with wrapper functions, but this pollutes the module namespace when the function would be otherwise unused:

```
struct Foo { bar: u32, baz: u32 }

fn get_bar(f: Foo) -> u32 { f.bar }

fn foo_sum(f: Foo) -> u33 { uadd(f.bar, f.baz) }

fn main() {
    let foos: Foo[u32:3] = [Foo { bar: 1, baz: 2 }, Foo { bar: 3, baz: 4 }, Foo { bar: 5, baz: 6 }];
    let bars: u32[u32:3] = map(foos, get_bar);
    let sums: u33[u32:3] = map(foos, foo_sum);
}
```

### Your view of the "best case XLS enhancement" (limit 100 words)

As a convenience, it'd be nice for map to support some syntactic sugar that parses as an anonymous function (even if we don't necessarily want to support them in general contexts), e.g.

```
struct Foo { bar: u32, baz: u32 }

fn main() {
    let foos: Foo[u32:3] = [Foo { bar: 1, baz: 2 }, Foo { bar: 3, baz: 4 }, Foo { bar: 5, baz: 6 }];
    let bars: u32[u32:3] = map(foos, | x | -> x.bar );
    let sums: u33[u32:3] = map(foos, | x | -> uadd(x.bar, x.bar));
}
```

mikex-oss, 2024-11-13:
Another example where a top-level function isn't possible is writing a [closure](https://doc.rust-lang.org/book/ch13-01-closures.html) where we need to freeze some arguments using local scope.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @erinzmoore): DSLX now supports anonymous functions (lambdas `|x| ...`) in `map()` via `class Lambda` (`xls/dslx/frontend/ast.h:3825`), `Parser::ParseLambda` (`xls/dslx/frontend/parser.cc:315`), and `RewriteLambdas` (`xls/dslx/frontend/lambda_rewriter.h:43`), tested in `xls/dslx/tests/lambda.x` and `xls/dslx/ir_convert/ir_converter_lambda_test.cc`. (Closed by Gemini)

closed 2026-11-14 by allight