google/xls #1555: [enhancement] DSLX should not care about function definition order as long as visible in some scope at invocation

mikex-oss, 2024-08-23:
### What's hard to do? (limit 100 words)

Unlike Rust (or even Python), DSLX requires functions to be defined above first use.

Borrowing from https://doc.rust-lang.org/book/ch03-03-how-functions-work.html, this does not work:

```
fn main() -> () {
    trace_fmt!("Hello, world!");
    another_function();
}

fn another_function() { trace_fmt!("Another function."); }
```

```
0001: fn main() -> () {
0002:     trace_fmt!("Hello, world!");
0003:     another_function();
~~~~~~~~~~^--------------^ ParseError: Cannot find a definition for name: "another_function"
```

This should work as long as `another_function` is defined in some scope visible to the body of `main`.

### Current best alternative workaround (limit 100 words)

You must define the functions in order:

```
fn another_function() { trace_fmt!("Another function."); }

fn main() -> () {
    trace_fmt!("Hello, world!");
    another_function();
}
```

### Your view of the "best case XLS enhancement" (limit 100 words)

Seems we need a first pass to collect the functions defined in scope.

proppy, 2024-08-23:
related/dup of https://github.com/google/xls/issues/197 ?

mikex-oss, 2024-08-23:
Duplicate of #197