google/xls #1568: [enhancement] DSLX bit type indexing

mikex-oss, 2024-08-28:
### What's hard to do? (limit 100 words)

If I try to write:

```
let foo: u23 = u23:0b10;
let bar = foo[1]; // expect u1:1
```

I get:

```
TypeInferenceError: uN[23] Value to index is not an array.
```

### Current best alternative workaround (limit 100 words)

You need to do a roundabout bit slice instead.

```
let bar = foo[1+:u1];
```

### Your view of the "best case XLS enhancement" (limit 100 words)

Support indexing, like we have in arrays. Note that this also works in Verilog and Python (I only reference Python because [the docs do](https://google.github.io/xls/dslx_reference/#bit-slice-expressions:~:text=DSLX%20supports-,Python%2Dstyle,-bit%20slicing%20over)).

cdleary, 2024-09-03:
One of the reasons this was left out historically is that bit vectors, perhaps confusingly, index differently from arrays, and we'd be unifying the notation; e.g.

```dslx
fn main() {
    let x = u3:0b001;
    trace_fmt!("{:b}", x[0 +: u1]);  // gives the least significant bit, which is right-most in the above, i.e. 1
    let y = [false, false, true];
    trace_fmt!("{:b}", y[0]);  // gives you the first item, which is left-most in the above, i.e. 0
}

#[test]
fn test_main() { main(); }
```

`bazel run -c opt //xls/dslx:interpreter_main  -- /tmp/foo.x  --alsologtostderr`

If `x[0]` gave you `1` and `y[0]` gave you `0` it seemed like it'd be confusing, at least the syntactic difference of a width-slice or a `[from:to]` slice seemed to help indicate the indexing would happen differently from an array.

mikex-oss, 2024-09-03:
Maybe this is some hardware background leaking through, but that all seems fine... In SystemVerilog, you could have:

```
logic foo = 3'b001;
logic bar[3] = '{1'b0, 1'b0, 1'b1};
```

where `foo[0] == 1'b1` and `bar[0] == 1'b0`.

mikex-oss, 2024-10-31:
Re-raising this as it's continued to be a regular mistake writing DSLX.

Take the example of related bit vectors, one for data and one for valid:

```
data: bits[DATA_WIDTH][N]
valid: bits[N]
```

Now when indexing into these, you end up with

```
data[i]
valid[i+:u1]
```

whereas in SystemVerilog, you really never think about this distinction when you write:

```
logic [N-1:0][DATA_WIDTH-1:0] data;
logic [N-1:0] valid;
```

Why not use `bool[N]`, i.e. `bits[1][N]`, which keeps the usage consistent? Arrays of single bits have other ergonomic issues, e.g. missing first-class and/or reduction of arrays (#331).