google/xls #1571: interpreter_main compare should default to none

proppy, 2024-08-29:
**Describe the bug**

By default `interpreter_main` set `--compare` to `jit` which end up compare every test result of the dslx interpreter w/ the jit'ed version of the same function.

While this is useful when working on the toolchain to verify that a given change has no impact on the JIT (or ir conversion), in practice this has little value to the developer as they are likely to only care about one result.

**To Reproduce**
```
bazel test //xls/examples/...
```

**Expected behavior**

The default should be `--compare=none` for both `interpreter_main` and `dslx_test` bazel rule, so that it only run the tests thru the interpreter.

mikex-oss, 2024-08-29:
FYI, the bazel rule is already being changed for #1415 

> Change default comparison method to none to avoid confusing errors when multiple procs are defined in a test.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Both `xls/dslx/interpreter_main.cc:74` (`ABSL_FLAG(std::string, compare, "none", ...)`) and `xls/build_rules/xls_dslx_test.bzl:36-38` (`_DEFAULT_DSLX_TEST_ARGS = {"compare": "none"}`) now default to `compare=none`. (Closed by Gemini)

closed 2026-11-14 by allight