google/xls #1572: Parametric signedness bit type doesn't work

mikex-oss, 2024-08-29:
**Describe the bug**
The bytecode emitter errors out if you try to use a return type like `xN[RESULT_SIGNED][RESULT_SZ]`.

```
: INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: Bits-to-array cast bit counts must match. Saw 32 vs 0.; while evaluating: to_int<u32:8, u32:23, u32:32>(APFloat<u32:8, u32:23> { sign: u1:0, bexp: u8:0x0, fraction: u23:0x0 })
...
Error: INTERNAL: Bits-to-array cast bit counts must match. Saw 32 vs 0.; while evaluating: ...
=== Source Location Trace: ===
```

**To Reproduce**
Steps to reproduce the behavior:
Change this line:

https://github.com/google/xls/blob/69f00f93eb09c0912185ac8a95b8075cb62a7463/xls/dslx/stdlib/apfloat.x#L1301

to `xN[RESULT_SIGNED][RESULT_SZ]` (updating the casts at https://github.com/google/xls/blob/69f00f93eb09c0912185ac8a95b8075cb62a7463/xls/dslx/stdlib/apfloat.x#L1352 and https://github.com/google/xls/blob/69f00f93eb09c0912185ac8a95b8075cb62a7463/xls/dslx/stdlib/apfloat.x#L1355)

**Expected behavior**
This should properly include parameterized signedness and remove the need to cast the result in the wrapper at https://github.com/google/xls/blob/69f00f93eb09c0912185ac8a95b8075cb62a7463/xls/dslx/stdlib/apfloat.x#L1370

mikex-oss, 2024-08-29:
@proppy suggested this undocumented feature, pointing to https://github.com/google/xls/commit/9761efba8421812abbfd8ef78c789469bf5e3dd9.

Once fixed, it would also be good to document it at https://google.github.io/xls/dslx_reference/#bit-type. `xN` is referenced in a couple places, but it's not clear this is not just a convenience shorthand for documentation.

cdleary, 2024-09-13:
Note this test case exhibits the same problem:

```
#[test]
fn casts() {
    assert_eq(s1:1, u1:1 as xN[true][1]);
    assert_eq(u1:1, s1:1 as xN[false][1]);
}

#[test]
fn widening_casts() {
    assert_eq(s2:0b11, s1:1 as xN[true][2]);
    assert_eq(u2:0b01, u1:1 as xN[false][2]);

    assert_eq(s2:0b11, xN[true][1]:1 as s2);
    assert_eq(u2:0b11, xN[false][1]:1 as u2);
}
```

PR #1606 (2024-09-20): [DSLX] Fixes for casting to/from `xN` (bits constructor) based type, adds docs and more tests
Fixes #1572 

Development details -- note that `xN` is the first place where the concept of "BitsLike" became load bearing, any place that assumed `ArrayType` in the hierarchy now has to consider that `ArrayType` could be indicating something that's effectively a bits type; i.e. `BitsLike`. Going forward we always want to use `BitsLike` as the test for a type and whether it's effectively a bits type, there were a few spots here that were missed when the `xN` bits type constructor was introduced.
closed 2024-09-20 by copybara-service[bot]