google/xls #1575: Make it possible to choose llvm toolchain to come from local system

hzeller, 2024-08-30:
Currently, the pre-packaged toolchain does not work on all systems, for instance @cdleary has to remove it to compile xls locally on his machine:

 * https://github.com/google/xls/commit/76d4ad5524dc5207f45c85f1498c445ffb2d6b99
 * Also @proppy had to do that to compile on the Mac: https://github.com/google/xls/pull/1570#issuecomment-2320601018

If there is a sufficiently compatible toolchain on the system (like in the cases above), it would be good if these users can opt out without having to patch  out the toolchain first. Maybe we need to define an option in bazael e.g. `//xls_use_system_compiler` that we then can select on to not load our toolchain. Not sure how that would be done, but it is probably is not terribly complicated either.
Should be tackled once #1570 is in.

hzeller, 2024-09-03:
Ok, #1570 is postponed, but switching toolchains makes sense at current head as well. If it is a lot of work, we should first gauge priority.

cdleary, 2024-09-04:
As you mentioned I do this just by hacking it out in the release branch. I did try to find a more proper way of doing it back in May but there's some fetch/initialize order dilemmas from what I could tell, I think I ended up down a rabbit hole trying to make custom starlark rules so I could either register- or not-register a toolchain. Somebody with better Bazel skills may be able to do this much more easily, I ended up giving up on trying for the proper way.