google/xls #1582: [enhancement] Add test points for mixed width operands in the delay model

rw1nkler, 2024-09-03:
### What's hard to do? (limit 100 words)

Currently, DSLX forces the user to cast operands of the arithmetic operations to the same type. However, for some binary operations the underlying original width of the operand is preserved in the IR, and is used in Verilog code to perform operation on the original operand widths, enabling the synthesis tool to create smaller hardware when possible.

The same type of optimization seems to be impossible on the scheduler level, as most of the test data points for the delay model use the same widths of operands. This will definitely affect operations like multiplication, but possibly other operations may benefit from that too (see [LINK](https://github.com/google/xls/blob/main/xls/delay_model/samples.textproto#L1579))

### Current best alternative workaround (limit 100 words)

To obtain more accurate timing information for the proc, one can use the actual P&R data. However, there is no way to leverage the problem with "sub-optimal" delay model for the scheduler

### Your view of the "best case XLS enhancement" (limit 100 words)

Extend the characterization process by providing more test points for the delay model

rw1nkler, 2024-09-03:
While working on opening this issue, I noticed a small discrepancy between handling multiplication and division in XLS. Below, there are two functions showing the difference.

##### Multiplication

* DSLX 

```rust
fn mul1(x: u32, y: u8) -> u32 {
    x * y as u32
}
```
  * IR
  
```
top fn __mul__mul1(x: bits[32], y: bits[8]) -> bits[32] {
  ret umul.8: bits[32] = umul(x, y, id=8, pos=[(0,3,6)])
}
```
  
  * Verilog
```verilog
module mul1(
  input wire [31:0] x,
  input wire [7:0] y,
  output wire [31:0] result
);
  // lint_off MULTIPLY
  function automatic [31:0] umul32b_32b_x_8b (input reg [31:0] lhs, input reg [7:0] rhs);
    begin
      umul32b_32b_x_8b = lhs * rhs;
    end
  endfunction
  // lint_on MULTIPLY
  wire [31:0] umul_11;
  assign umul_11 = umul32b_32b_x_8b(x, y);
  assign result = umul_11;
endmodule
```

It can be seen that the original widths of the operands were preserved in both IR and the generated Verilog (in the function that wraps the operation).

##### Division

  * DSLX 
```rust
fn div1(x: u32, y: u8) -> u32 {
    x / y as u32
}
```
  * IR
```
top fn __mul__div1(x: bits[32], y: bits[8]) -> bits[32] {
  literal.5: bits[24] = literal(value=0, id=5)
  concat.6: bits[32] = concat(literal.5, y, id=6)
  ret udiv.4: bits[32] = udiv(x, concat.6, id=4, pos=[(0,15,6)])
}
```
  * Verilog
```verilog
module div1(
  input wire [31:0] x,
  input wire [7:0] y,
  output wire [31:0] result
);
  function automatic [31:0] udiv_32b (input reg [31:0] lhs, input reg [31:0] rhs);
    begin
      udiv_32b = rhs == 32'h0000_0000 ? 32'hffff_ffff : lhs / rhs;
    end
  endfunction
  wire [31:0] udiv_12;
  assign udiv_12 = udiv_32b(x, {24'h00_0000, y});
  assign result = udiv_12;
endmodule
```

In the case of division, we can observe that the `udiv` uses operands of the same width in the IR, te same is true for the function that wraps the operation in Verilog. The division will be optimized by the synthesis tools anyway, but I wonder whether this difference in behavior was intentional in the toolchain.

proppy, 2024-09-04:
> However, there is no way to leverage the problem with "sub-optimal" delay model for the scheduler

@QuantamHD @richmckeever couldn't this also enabled by fdo?
https://github.com/google/xls/blob/f02c9999c3a739cbcbd6f414ea19fb10439f3cf1/xls/tools/scheduling_options_flags.proto#L47-L58