google/xls #1584: Build Failing after recent git pull

dassbrothers, 2024-09-04:
When I did the build last time in the first week of August the build was successful.
```
git pull 
bazel build -c opt xls/...
```
However, when I do the same now, I get the following error: **An error occurred during the fetch of repository 'llvm-project'**. I also tried cleaning the build and then rebuild from scratch as well, but that did not help as well. 

```
git pull
bazel clean
bazel build -c opt xls/...
```
**Environment:**
 - OS: [e.g. Ubuntu]
 - Versions [e.g. 22.04]
 - ldd version: Ubuntu GLIBC 2.35

I am attaching the screenshot of the same. 

[image]

proppy, 2024-09-04:
Looks like a timeout error, is that transient? do you have a reliable internet connection?

proppy, 2024-09-04:
I'm curious if you get the same error when trying out the same commands from a cloud environment (azure, aws or gcp)?

dassbrothers, 2024-09-04:
My internet connection is stable, and I have tried it multiple times over the past two days, but I am getting the same error again and again. How to run the commands from the cloud environment?

dassbrothers, 2024-09-18:
The build passed after removing everything from the "/home/sourav/.cache/bazel/_bazel_sourav/..." directory

closed 2024-09-18 by dassbrothers