google/xls #1591: Storing dslx::Pos with filename strings is pretty expensive

hzeller, 2024-09-05:
Currently, we're storing the full filename as a string in [`dslx::Pos`](https://github.com/google/xls/blob/7eb7d07b756df405398755734ff773c0061d881a/xls/dslx/frontend/pos.h#L89) (and two `dslx::Pos` in a [`dslx::Span`](https://github.com/google/xls/blob/7eb7d07b756df405398755734ff773c0061d881a/xls/dslx/frontend/pos.h#L156-L157)).

This is a lot of unnecessary memory that has to be moved around all over.

A quick test running one of the code examples

```
bazel-bin/xls/dslx/interpreter_main xls/dslx/stdlib/apfloat.x
```

This churns through about 6'000'000 Pos objects with > 100MiB of strings. Just for `dslx::Pos`.

In addition, all `dxls::Token`s [contain a string](https://github.com/google/xls/blob/7eb7d07b756df405398755734ff773c0061d881a/xls/dslx/frontend/token.h#L188), instead of a `std::string_view into` the original file. Which means it looses all context and requires for the token to store an additional span and pass around that as expensive tuple.

## To Improve

I suggest to have an object that represents a file, a tuple `(filename, content, content_offset_to_linenumber_index)`, and hold on to this for lifetime of a dslx parse. Now, everything becomes much simpler:

Tokens can just be string-views into the original content, Positions and Spans are not needed anymore and can be implicitly extracted from the `std::string_views` and can be mapped back to filename, line, column when needed on error reporting or similar (by consulting the one `(filename, content, index)` tuple) without having to keep one or multiple `(filename, line, column)` tuple at every token, Node, etc. 

In addition, the formatter can be simplified a lot, as it does not have to keep comments 'on the side', but can just point to the place where they actually are.

This will help dramatically to reduce unnecessary memory churn.

Category: cleanup and simplification. I don't have time to work on this currently.
If someone wants to work on it, I have some detailed thoughts and working framework that can be used as an example.

hongted, 2024-09-05:
Or in general, consider arena allocation for strings and other objects like nodes that are pseudo-static and used often.

cdleary, 2024-09-06:
I can for sure take on making a file table since this is my tech debt.

I intentionally had tokens hold strings (when they have to hold strings, ideally they'd mostly hold structured content instead) because then you don't need to worry about the "lifetime ownership" object for the token stream. For AST module is the arena which is pretty natural. For production frontends I'd worked on before keeping ASTs associated with in memory file contents is a bit of a smell because then you have to keep all the source text around in memory even if it's huge, so I'm not really bought in on that part, but making a file table would be easy and would reduce most of the cost cited in the OP.

hzeller, 2024-09-06:
I remember you mentioned the potential cost of keeping the file in memory before.
I'd say, the text content that you have to store once is always cheaper than creating a bunch of strings representing exactly their content (plus the malloc-overhead for small pieces), plus the now necessary additional metadata to keep track of positions etc. I'd say it saves at least 2-10x (and makes a lot of things much easier and faster). It 'feels' like it might use a lot of memory to keep the file with non-token whitespace around by keeping the file content, but since it is contiguous, and there are less complicated data structures to keesp on the side  it immediately saves. Depends on if early tokens have to be kept or can be discarded before the file is finished.

Anyway, I think the memory of tokens is not a super-important issue right now, but might be fun to work on at some point as I think it will also make the formatter-code a little easier to deal with. I  only mentioned it as with that, we wouldn't even need `dslx::Pos` and `dslx::Span` anymore, the starting point of this.

Coming back to the original Pos representation: a file-table and in the `dsls::Pos` just an index instead of the filename would already be a good start indeed and would match nicely how we refer to files in other contexts.

cdleary, 2024-09-10:
@hzeller I'd recommend we close this one and break out the "hold fewer strings" as a separate issue -- I think an interned string table for identifiers would be a good idea though we may want to motivate its relative importance with some profiling if possible. (This may be easier for you to do than me since I don't have an easy way to get a tcmalloc based build in the bazel repo.)

closed 2024-09-10 by hzeller