google/xls #1592: [enhancement] XLS proc jit should detect at least trivial livelock conditions

allight, 2024-09-05:
### What's hard to do? (limit 100 words)

If you run a jit until blocked we have deadlock detection but we do not have livelock detection of any kind. Even trivial livelocks like an empty proc will make the 'until-blocked' useless.

### Current best alternative workaround (limit 100 words)

Perform DFE to remove empty procs before running the proc jit and avoid other livelock patterns.

### Your view of the "best case XLS enhancement" (limit 100 words)

Detect at least trivial live-lock conditions and return errors if they are entered.

meheff, 2024-09-16:
Without solving the halting problem, one relatively easy case to catch would be: no data sent on channels, all procs blocked or iterating without changing state.

allight, 2024-09-16:
yeah that's basically what I'm thinking of. Issue is that right now we don't really keep track of the old state for anything.

Doing any sort of detection when values are actually being passed along channels is much more difficult.

And of course full livelock detection is halting problem.